neu-rah/ArduinoMenu #267: How can I use MCUFRIEND_kbv as an output for ArduinoMenu?

vikulin, 2019-12-07:
I have 3.5" UNO Module which I would like to use together with ArduinoMenu. I did find [MCUFRIEND_kbv](https://github.com/prenticedavid/MCUFRIEND_kbv ) library.
How can I use this lib?

Board: Arduino Uno

TFT spec:
Driver: ILI9486
Dots:320x480
Bus:8bit
Touch:No

neu-rah, 2019-12-07:
you can issue navigation commands to the navigation root instead on depending on full automation, this example is about it: https://github.com/neu-rah/ArduinoMenu/blob/master/examples/codeCtrl/codeCtrl/codeCtrl.ino

please note the usage of NONE on the input definition... only needed if using the macros (see nomacros example)

neu-rah, 2019-12-07:
sorry, you talking about input, right? using the display... it is compatible with adafruit GFX, therefor it could be used as so

vikulin, 2019-12-07:
Thanks for quick answer. GFX should be used with hardware specific lib.
Such as TFT or MCUFRIEND_kbv. I found an example for UTFT lib and some
more. I'm looking for an example for mentioned MCUFRIEND_kbv lib because
it's working for Uno board oppositely to UTFT which requires Mega.

neu-rah, 2019-12-07:
you are welcome, we have also a chat room https://gitter.im/ArduinoMenu/Lobby

if a the display driver is nor yet supported we can add a menu output driver for it, its not hard, menuIO folder contains some, usually i start with the most similar and adapt from it

NeftaliLoya, 2020-05-21:
Hello vikulin. If you are using the 3.5 inch ILI9846 TFT-LCD, just add the following to your sketch.

#define LCD_CS A3        // Chip Select goes to Analog 3
#define LCD_CD A2        // Command/Data goes to Analog 2
#define LCD_WR A1        // LCD Write goes to Analog 1
#define LCD_RD A0        // LCD Read goes to Analog 0
#define LCD_RESET A4  // Can alternately just connect to Arduino's reset pin
#include "Adafruit_GFX.h"
#include <TFT_HX8357GLUE.h>
TFT_HX8357GLUE tft;
#ifndef AVR
#include <avr/dtostrf.h>
#endif

The TFT_HX8357GLUE.h loads the MCUFIEND_kbs library for you and it is part of the MCUFRIEND_kbv-master download bundle.

This TFT-LCD was developed to be used as an Arduino UNO shield; therefore, the access to the on-board SD card is seamless on the UNO. If you want to use this TFT-LCD on the Arduino MEGA-2560 and on top of that you'd like to use the so mentioned on-board SD slot, you must use SoftSPI to bit-bang the none-ISP pins on the MEGA. To use this SoftSPI version, just add the following to your sketch:

#include <SPI.h>                      //  f.k. for Arduino-1.5.2
#define USE_SDFAT.                //  SDFAT allows the use of SoftSPI
#include <SdFat.h>                  //  Use the SdFat library
SdFatSoftSpi<12, 11, 13> SD;  //  Bit-Bang on the Shield pins
#define SD_CS 10                     //  Chip select for the SD

Evidently, do not forget to download and to install the required libraries.

Last, but not least, just initialize the TFT-LCD and the SD instances by adding the following to your code's void setup():
tft.begin();  
SD.begin(SD_CS);

These additions will allow you to use your TFT-LCD along with all the examples provided in the MCUFRIEND_kbv-master download package.

Then you can start building your menu by using the adafruitGfx/tft/tft.ino example that comes in the ArduinoMenu library bundle.

I'll be more than willing to help you if you need further assistance on using this TFT-LCD.

Best regards...