neu-rah/ArduinoMenu #268: M5StickC support?

gnikolopoulos, 2019-12-10:
Hi all,

I was wondering if there are plans to also provide support for using M5StickC's built in-display (SST7735S, 160x80). It's a neat device and being self-contained, makes it ideal for small projects.

Also, if anyone has started working on something like this,let me know, I will be glad to help...

neu-rah, 2019-12-10:
the esp32 has arduino framework  support and ST7735S has some drivers for adafruitGFX, so it should work. I wonder if UCGLib supporrt dor sST7755 can also work with the variant.

just some links, you may find some others
https://github.com/espressif/arduino-esp32

https://github.com/adafruit/Adafruit-ST7735-Library

https://github.com/olikraus/ucglib

gnikolopoulos, 2019-12-11:
Hi @neu-rah ,

I tried using UCGLib and it mostly worked, but the problem is that UCGLib is quite big and takes more than 60% of available program memory just for the demo alone... Adafruit Library did not work since the pins used on M5StickC are not the ones Adafruit library expects and there is no obvious way to change them during runtime...

Is there a detailed tutorial on how to create your own outputs? I know I can use others as reference, but a detailed explanation would help figure out what the minimum required functions are etc...

neu-rah, 2019-12-11:
Hi!  @gnikolopoulos 

not yet, usually i start by a similar driver from menuIO folder and adapt from it... but yest we need a tutorial on this, writing it now