neu-rah/ArduinoMenu #269: Cant loop!

mariusdp, 2020-01-01:
Hi all,

First I apologise if I am in the wrong place, please point me in the wright place if that is the case. The issue: I attache a second i2c oled and on it it displays value read from analogue pin in "drawA, B or C". When I press on "ADC12" in "Voltmetre" menu the data is displays on the second oled but only one time (does not update/loop). The question is how can I do that??? Tried all sort of suggestions found out there but with no success. If someone can help that would be great. 

THANK YOU,

`/********************
Arduino generic menu system
lolin32 menu example

output: onboard oled (i2c ssd1306 u8g2) + Serial
input: Button + Serial
mcu: wemos lolin32 (ESP32) with builtin oled

********************/
#include <Arduino.h>
#include <Wire.h>
#include <menu.h>
#include <menuIO/u8g2Out.h>
#include <menuIO/serialIO.h>
#include <menuIO/stringIn.h>
#include <menuIO/chainStream.h>
#include "AiEsp32RotaryEncoder.h"
// #include <menuIO/encoderIn.h>

// Define LEDPIN.
#define LEDPIN 14

#define MAX_DEPTH 2

// Define buttons.
#define ROTARY_ENCODER_A_PIN 0
#define ROTARY_ENCODER_B_PIN 2
#define ROTARY_ENCODER_BUTTON_PIN 4
#define ROTARY_ENCODER_VCC_PIN -1 /*put -1 of Rotary encoder Vcc is connected directly to 3,3V; else you can use declared output pin for powering rotary encoder */

// Define buttons.
// #define NAV_BTN 0 // DOWN button
// #define NAV0_BTN 2  // UP button.
// #define SEL_BTN 4 // SELECT button

using namespace Menu;

#define SDA 5
#define SCL 4

#define fontName u8g2_font_7x13_mf
#define fontX 7
#define fontY 16
#define offsetX 0
#define offsetY 3
#define U8_Width 128
#define U8_Height 64
#define USE_HWI2C
#define fontMarginX 2
#define fontMarginY 2

U8G2_SSD1306_128X64_VCOMH0_F_HW_I2C u8g2(U8G2_R0, U8X8_PIN_NONE, SCL, SDA);//allow contrast change
U8G2_SSD1306_128X64_VCOMH0_F_HW_I2C u8g2a(U8G2_R0, U8X8_PIN_NONE, SCL, SDA);//allow contrast change

AiEsp32RotaryEncoder rotaryEncoder = AiEsp32RotaryEncoder(ROTARY_ENCODER_A_PIN, ROTARY_ENCODER_B_PIN, ROTARY_ENCODER_VCC_PIN);
int test_limits = 10;

//+++++++++++ Voltmetre variables ++++++++++++++
float vout = 0.0;
float vin = 0.0;
int value = 0;
//**********************************************

/************************ 
[... 365 lines cut ...]
  // if (!digitalRead(NAV_BTN)) {
  //   delay(SOFT_DEBOUNCE_MS);
  //   while(!digitalRead(NAV_BTN));//wait for button release
  //   nav.doNav(upCmd);
  //   delay(SOFT_DEBOUNCE_MS);
  // }
  // if (!digitalRead(NAV0_BTN)) {
  //   delay(SOFT_DEBOUNCE_MS);
  //   while(!digitalRead(NAV0_BTN));//wait for button release
  //   nav.doNav(downCmd);
  //   delay(SOFT_DEBOUNCE_MS);
  // }
  nav.doInput();
  digitalWrite(LEDPIN, !ledCtrl);//no led on this board
  if (nav.changed(0)) {//only draw if menu changed for gfx device
    u8g2.firstPage();
    do nav.doOutput(); while(u8g2.nextPage());
  }
  
  delay(100);//simulate other tasks delay
  rotary_loop();
  delay(50);															 
	if (millis()>20000) rotaryEncoder.enable ();

}`

neu-rah, 2020-01-02:
Hi! this is the right place, welcome

on my first look on your code:

you have one automated input `serialIn serial(Serial);`

and two menu outputs, 
```c++
//define outputs controller
menuOut* outputs[]{&outSerial,&oledOut};//list of output devices
outputsList out(outputs,sizeof(outputs)/sizeof(menuOut*));//outputs list controller
```
on the loop:
`if (nav.changed(0)) {`
this will check if check if visible menu on device index 0 has changed and therefor needs to be drawn, on your case device index 0 is `&outSerial` (not the gfx) **but that should work**

my question is, does your code work if you remove all the extra rotary code and delays and use serial as sole input instead?  please use keys `+` `-` `*` `\` followed by enter to navigate menu on serial monitor

mariusdp, 2020-01-02:
Thank you for looking in! Yes it works, started like that and added rotary after. The "result drawA" could never get it to loop. Tried normal delay(xxx); if millis xxx do... Even tried to use two oled drivers hopping that "draw" would be more independent - result was the same (was interesting to see it displaying menu with u8g2 and draw with adafruit_ssd1306). thank you!

mariusdp, 2020-01-02:
PS: navigation works fine on all 3 way tried so far: serial, encoder and btn. Could never get it to work with your encoder though, that is why I added that encoder bit.

neu-rah, 2020-01-03:
the builtin encoder was done and tested for avr's, it depends on PCINT library that eventually can work on other platforms, but not tested.
There is also clickencoder driver, the original as you can see here https://github.com/0xPIT/encoder is also for AVR's but its because of its timer1 dependency and some includes that can be discarded, I've used this driver with an esp8266 just by replacing the timer and removing avr specific includes. I can provide esp8266 code fot that.
Since AM4 now has code driven navigation we can adapt any input driver by just calling the navigation functions... and after that didn't worry much on providing input specif drivers...

now having a second display to for independent draw, since you want results to be fixed and updating i would use menu handler to set a drawing type flag only and then I would do an independent switch/case (inside loop) to draw updated info on the second screen and it will keep drawing updated info according to selected flag value. This is the only way to have parallel draw and free the menu for navigation

mariusdp, 2020-01-03:
1. "now having a second display..." can you point me towards an example on how to achieve this? I´m kind of new to this world :P

2. The code for esp8266 with 0xPIT/encoder would be great.

Thank you,