neu-rah/ArduinoMenu #270: Is-this-Submenu-focused? or inside-this-submenu-page?  function.

HamidSaffari, 2020-02-15:
Hi there.
Is there any way that we realize which submenu we are at now, so in the loop() we can write something to show when we navigated inside that special submenu?
something like:
if ( mainMenu[2].isFocused() ) { showTime() }

neu-rah, 2020-02-15:
Hi!

yes there is, you can either

1) provide focus handler functions
2) inspect NavRoot (nav) object members  
  - `nav.level` indicates the depth level
  - `nav.path[nav.level].sel` the current selected index

member path is an array of `NavNodes` and it describes the navigation way up to the current `level`, you can also thing of it in reverse order, it provides the current menu (nav.path[nav.level]) and all  parent's until the root `nav.path[0]`

NavNode provides selection index `sel` as well as a pointer to the current associated menu `target` and a pointer to the root object.

HamidSaffari, 2020-02-15:
Thank neu-rah for respond.
but still can not get the unique number of special function that I want. if I use:
if( nav.level==2  &&  nav.path[nav.level].sel==1 ) do_something();
still referred to many sub-menus with select item number 1 inside.

neu-rah, 2020-02-15:
you're welcom
following that line, `nav.path[level].target` is the pointer to that level menu, you can compare.
still one can add handlers...

HamidSaffari, 2020-02-17:
so now how could I define my intended Sub-Menu to compare with the target pointer?
for example in this code how can I know I'm inside the chooseMenu?

TOGGLE(ledCtrl,setLed,"Led: ",doNothing,noEvent,noStyle//,doExit,enterEvent,noStyle
  ,VALUE("On",HIGH,doNothing,noEvent)
  ,VALUE("Off",LOW,doNothing,noEvent)
);
int selTest=0;
SELECT(selTest,selMenu,"Select",doNothing,noEvent,noStyle
  ,VALUE("Zero",0,doNothing,noEvent)
  ,VALUE("One",1,doNothing,noEvent)
);
CHOOSE(chooseTest,chooseMenu,"Choose",doNothing,noEvent,noStyle
  ,VALUE("First",1,doNothing,noEvent)
  ,VALUE("Second",2,doNothing,noEvent)
);
MENU(subMenu,"Sub-Menu",doNothing,noEvent,noStyle
  ,OP("Op",doNothing,noEvent)
  ,EXIT("<Back")
);
MENU(mainMenu,"Main menu",doNothing,noEvent,wrapStyle
  ,SUBMENU(subMenu)
  ,SUBMENU(setLed)
  ,SUBMENU(selMenu)
  ,SUBMENU(chooseMenu)
);

neu-rah, 2020-02-17:
Hi!
added example for event inspection https://github.com/neu-rah/ArduinoMenu/blob/master/examples/handlers/handlers/handlers.ino

it used event handler and mask, here using a shared event inspection function and `anyEvent` mask. I recommend you to use an individual handler on the specific menus you want to monitor and eventually a more restrictive mask, event masks can be combined to catch multiple events.

if you want to share the same handler with multiple menus, then you must use `target` pointer to compare with your menu and disambiguate between them, ex: `nav.target==&chooseMenu`

note that enumerated fields CHOOSE, SELECT, TOGGLE act like menus event even thou TOGGLE does not retain focus one can get the target variable value.

HamidSaffari, 2020-02-17:
Thank You very much, that example helped me a lot.
I use this condition in my loop:

void loop(){
...
	if ( nav.path[0].sel==2 && nav.level==1 && nav.sleepTask!=inaction ){
		lcdPrintTime();   }
}
//====================================================
and it works perfectly fine when you now the order of your menus items and level.
keep it coming neu-rah.

closed 2020-02-17 by neu-rah