neu-rah/ArduinoMenu #273: Adafruit_GFX custom font print text does not erase the background

HamidSaffari, 2020-03-08:
Hi, there.
When using bigger LCD or just want to use bigger font you need fonts with better resolution and only by using "tft.setTextSize(4);" you just get bigger font with that old resolution which is kinda ugly.
so when I use custom font made by Adafruit like this:
#include <Fonts/FreeMono24pt7b.h>
the resolution goes up but there is big problem and that is because some of these fonts are not standard fix character size so Adafruit do not implement erase-the-previous-character-code for them. the description on the Adafruit_GFX.cpp line 1199:
"
    // NOTE: THERE IS NO 'BACKGROUND' COLOR OPTION ON CUSTOM FONTS.
    // THIS IS ON PURPOSE AND BY DESIGN.  The background color feature
    // has typically been used with the 'classic' font to overwrite old
    // screen contents with new data.  This ONLY works because the
    // characters are a uniform size; it's not a sensible thing to do with
    // proportionally-spaced fonts with glyphs of varying sizes (and that
    // may overlap).  To replace previously-drawn text when using a custom
    // font, use the getTextBounds() function to determine the smallest
    // rectangle encompassing a string, erase the area with fillRect(),
    // then draw new text.  This WILL infortunately 'blink' the text, but
    // is unavoidable.  Drawing 'background' pixels will NOT fix this,
    // only creates a new set of problems.  Have an idea to work around
    // this (a canvas object type for MCUs that can afford the RAM and
    // displays supporting setAddrWindow() and pushColors()), but haven't
    // implemented this yet.
"
So when I use font for Menu the character write on top of previous char.
Now what do you prefer to do?
Putting getTextBounds() and fillRect() inside adafruitGfxOut.h can do the job?

neu-rah, 2020-03-08:
Hi!

you can adjust character dimensions when setting up the menu output, however things can get ugly when using non mono-space fonts, to get around this on u8g2 I've removed the print single char from the menu u8g2 output driver, maybe we need to do this for adafruit's

we gain the ability to deal better with unicode special characters (multi-byte chars) but lost the ability to contain text inside panel.. a fair trade i think

menu gfx drivers often use a draw rectangle before drawing.

HamidSaffari, 2020-03-11:
Adafruit has some mono-space custom font series with Freemonoxx name like FreeMono18pt7b.
Adafruit_GFX use a draw rectangle before drawing for his original font (gfxfont.h) only.
I think we could use what you used for Ucglib in UCGLibOut.h for adafruitGfxOut too so it wont write on top of the background text when using Freemono font.
beside I used Ucglib instead of Adafruit_GFX but the result was too slow in refreshing screen when navigating thru menu even with the same SPI clock frequency.

HamidSaffari, 2020-03-11:
I realized you did clear the the old character place when about the write new one in adafruitGfxOut.h as well.
oh I think the problem is cumming from the position for the character output.
as it starts from top for built-in font but Baseline for custom fonts.
but how can we change Baseline to top-left position in adafruit? like what Ucglib has: ucg.setFontPosTop();

 if (!gfxFont) { // 'Classic' built-in font

}else{ // custom font

}

HamidSaffari, 2020-03-13:
I did correct that by adding some code in Adafruit_GFX.cpp inside Adafruit_GFX::drawChar() in line 1197 :
if(_setFontPosTop){
x = x-xo;
y = y-yo;
}

and edit:
setFont(const GFXfont *f = NULL , bool setFontPosTop =true);
void Adafruit_GFX::setFont(const GFXfont *f, bool setFontPosTop) {
......
.....
_setFontPosTop = setFontPosTop;
}
bool _setFontPosTop;
//=====================================================

and it works but the lower case characters stuck to top line and the result is a bit ugly.

neu-rah, 2020-03-14:
Hi!
I will do some parallel things to u8g2 because it had the same problem, it can be corrected on menu output driver (i think).

u8g2 report here
https://github.com/neu-rah/ArduinoMenu/issues/185

at least the non-monospace problem along with UTF8 multibyte chars was solved here https://github.com/neu-rah/ArduinoMenu/issues/185 (for u8g2)

occamsshavingkit, 2022-10-04:
Hello! I had a related problem and found the reason: the cursor position on the default font is the top left corner of the first letter. The cursor position on custom fonts is the left baseline of the first letter. (https://learn.adafruit.com/adafruit-gfx-graphics-library/using-fonts) A solution for the menu system would involve calling getTextBounds to know where to draw the background. I unfortunately get a bit lost in the code.