neu-rah/ArduinoMenu #276: SDMenu plugin in WebMenu

RCP1, 2020-03-20:
Hi,

it seems the SDMenuT::printTo method was never updated for usage with XML translated HTML. 

After adding the additional wraps in the method, the menu shows up but as soons as one selects a file, the menu does not go back to the menu, where the CachedSDMenu was added nor updates the selected file.

neu-rah, 2020-03-21:
your file looks ok, so i've pasted the content over the original and loaded SDCard example on the due it looks ok either with simple serial output as with xmlFmt (good job!), so I think that youe issue is somewhere else...

my changes to use the existing example with xml output over serial port

```c++
//define input device
serialIn serial(Serial);

//define output device
idx_t serialTops[MAX_DEPTH]={0};
xmlFmt<serialOut> outSerial(Serial,serialTops);

//define outputs controller
menuOut* constMEM outputs[] MEMMODE={&outSerial};//list of output devices
outputsList out(outputs,1);//outputs list controller

// MENU_OUTPUTS(out,MAX_DEPTH
//   ,SERIAL_OUT(Serial)
//   ,NONE//must have 2 items at least
// );
//
// serialIn serial(Serial);
NAVROOT(nav,mainMenu,MAX_DEPTH,serial,out);
```

RCP1, 2020-03-21:
Yeah, i rather think that it is somewhere in the doNav function of SDMenu, since it outputs /displays correctly but does not go back to the original menu after selecting a file.

RCP1, 2020-03-22:
So, it seems to be a missing extension of SDMenuT. You use for web input the menuNode::async method, which calls the doNav method of SDMenuT, but with idxCmd instead of enterCmd? Hence, the file selection is never called via web async input? But even if I add idxCmd as additional case in SDMenuT::doNav, the selection index seems to be random...