neu-rah/ArduinoMenu #277: [Q] Delay between TOGGLE

leikoilja, 2020-03-22:
Hey all. 
Thank you for the amazing library.

I have tried finding it in documentation, but don't seem to find anything about how to make a slight delay between TOGGLE options? 

My problem is that when i press the button it changes the state like 5 times while i hold the button. A slight delay between button press of like 5-10ms would perfectly work here i guess. Instead of me introducing a custom delays is there any configuration option for that?

neu-rah, 2020-03-22:
Hi!

I avoid that by limiting the times that `.poll` is called per second

10 times = 100ms delay

leikoilja, 2020-03-22:
Ou, that's a good suggestion. Thank you, @neu-rah. 
Could you please share a quick reference how to do it?

neu-rah, 2020-03-22:
of course, ex: inside loop doing a minimal delay of 100ms, this wont be correct near millis overflow (that happens about every 5 days, but you get the point)

```c++
static unsigned int next=0;
if (millis()>=next) {
  nav.poll();
  next+=millis()+100;
}
```

_please note that I've not compiled this_

leikoilja, 2020-04-08:
Thank you, @neu-rah, for help! :+1:

closed 2020-04-08 by leikoilja