neu-rah/ArduinoMenu #278: Encoder error: "ISR not in IRAM" on Wemos D1 Mini

aliusc, 2020-03-23:
Hello,

I trying to set up this library with my hardware. I use Wemos D1 Mini, rotary encoder, 1602 LCD via PCF8574. For ideal test condition I selected LCD > PCF8574 example. 

I have wired encoder's CLK and DT to D5, D6, encoders switch to D7. I use ESP8266 board library 2.6.3. Code compiles well and starts up, but frequently reboots with Error in console `ISR not in IRAM!
`
When I comment out encoder and leave LCD and button only it starts up correctly. I have read interrupt functions should have attribute `ICACHE_RAM_ATTR ` to make them work correctly. But I have difficulties to add them to proper functions. I have tried to add it to "menuIO/encoderIn.h" to functions "encodeInUpdateA" and "encoderInUpdateB" declarations:
`ICACHE_RAM_ATTR static void encoderInUpdateA(class encoderIn<pinA,pinB> *e);
      ICACHE_RAM_ATTR static void encoderInUpdateB(class encoderIn<pinA,pinB> *e);`

and `ICACHE_RAM_ATTR void encoderIn<pinA,pinB>::encoderInUpdateA(class encoderIn<pinA,pinB> *e)` function declaration. Sadly it didn't helped. Maybe anybody had similar issue and may suggest where to add "ICACHE_RAM_ATTR" attribute? Thank you

neu-rah, 2020-03-23:
Hi!
The builtin encoder uses pin change interrupts and was tested on AVR's only, clickEncoder (an alternative) has been used with esp8266.

One can even use other encode/input method and issue menu commands when appropriate, after reading the input.

`codeCtrl` example is about menu commands

aliusc, 2020-03-26:
Saldy clickEncoder did not compiles for my board, but I'm not sad as I found `codeCtrl` example! It opens all possibilities of menu I need. I can control rotary encoder and button from my code. So i will adopt my project to work with this menu and get rid my old menu code I used as it is bit hardly maintained. Thank you for your support.

closed 2020-03-26 by aliusc