neu-rah/ArduinoMenu #280: Issue with ESP32 Clickencodertft.ino

timmcarson, 2020-03-27:
I am trying to use the ESP32 Clickencodertft.ino example from #256 

I am using PlatformIO running on vscode.  when I try to compile I get errors surrounding the ClickEncoder library.

This is what the errors look like
[image]

Any help is appreciated

Timm

neu-rah, 2020-03-27:
Hi!

Thanks for reporting it.

reorder could be  a warning or silenced and it would still work, that initialization is not order dependent. I will try to put it on proper order using vscode or activate the warning on atom+platformio....

meanwhile you might want to look at codeCtrl example (https://github.com/neu-rah/ArduinoMenu/issues/278#issuecomment-604662371) it allow you to use any input method and dont stay dependent of clickEncoder.

neu-rah, 2020-03-27:
Please note that order error are from clickEncoder lib, not that is wrong, it should be a warning at most, but your compile options vscode+platformio are adding an `-Werror=order` into the compiler options.

My atom+ platformio are also doing the same for esp32. I was able to compile by disabling it as an error and also added some warning suppress for a quiet compile (optional)

added this line to platformio.ini
```ini
build_flags = -Wno-comment -Wno-reorder -Wno-strict-aliasing -Wno-builtin-macro-redefined -Wno-switch
build_unflags = -Werror=reorder
```

timmcarson, 2020-03-27:
Adding the lines to the platformio.ini file did the trick it compiles now. 

Thanks for the quick response

neu-rah, 2020-03-27:
https://github.com/espressif/arduino-esp32/issues/3854

closed 2020-03-30 by timmcarson