neu-rah/ArduinoMenu #283: Change Font Size?

timmcarson, 2020-04-06:
I have successfully compiled and ran the esp32 clickEncodertft example.  I know want to change the font size.  changing fontW and fontH changes the size of the display lines but the not font size itself.  How do I change the font type and the font size?

Thanks

smp4488, 2020-04-07:
@timmcarson take a look here and see how I was able to do it https://github.com/smp4488/Guitarix-Pedalboard/blob/master/src/main.cpp#L302

mikejr83, 2024-01-21:
@timmcarson were you able to solve this? I'm attempting to do the same on an OLED using the Adafruit GFX out output. Trying the sam ething as @smplman didn't get me anywhere.

```c++
#define FONT_SCALE 2
MENU_OUTPUTS(out,
             MAX_DEPTH,
             ADAGFX_OUT(oled, colors, FONT_SCALE * 6, FONT_SCALE * 9, {0, 0, SCREEN_WIDTH / FONT_SCALE * 6, SCREEN_HEIGHT / FONT_SCALE * 9}),
             SERIAL_OUT(Serial));
```
This essentially just changes the line size but not the font size. It appears to be what @smplman is doing.