neu-rah/ArduinoMenu #284: Skip button click. Example TFT_eSPI/ArduinoMenu_LilyGo_TTGO_T-display_demo

apronin83, 2020-04-09:
To solve the problem, use this code

`void button_init()
{
    btnUp.setLongClickHandler([](Button2 &b) {
        unsigned int time = b.wasPressedFor();

        if (time >= 1000) {
          // Select
          nav.doNav(enterCmd);
        } else {
          // Up
          **nav.doNav(downCmd);**
        }
    });

    btnUp.setClickHandler([](Button2 &b) {
       // Up
       nav.doNav(downCmd); // It's called downCmd because it decreases the index of an array. Visually that would mean the selector goes upwards.
    });

    btnDwn.setLongClickHandler([](Button2 &b) {
        unsigned int time = b.wasPressedFor();
        
        if (time >= 1000) {
          // Exit
          nav.doNav(escCmd);
        } else {
          // Down
          **nav.doNav(upCmd);**
        }
    });

    btnDwn.setClickHandler([](Button2 &b) {
        // Down
        nav.doNav(upCmd); // It's called upCmd because it increases the index of an array. Visually that would mean the selector goes downwards.
    });
}`

neu-rah, 2020-04-11:
"To solve the problem" .... something tells me that this should be a response to other issue and not an issue by itselff, am I right?

cmeldas, 2020-04-16:
I suppose the issue is, that:
Long click in Button2.h is by default 200ms and more.
In code above, long click i filtered for 1000ms.
So there is gap, 200 to 1000ms, when is no action.

This behavior i can confirm.

**This is my suggested fix**
,but required latest Button2. Recently was in Button2 accepted this change: https://github.com/LennartHennigs/Button2/issues/10
Because of this change we can do this:

At the beginning:
```
#define LONGCLICK_MS    500  //1000ms is too long for me, default 200ms is to short
#include <Button2.h>
```

So button_init function is more strait and we can change longpress duration as we want.
```
void button_init()
  {
    btnUp.setLongClickHandler([](Button2 &b) {
        nav.doNav(enterCmd);
    });

    btnUp.setClickHandler([](Button2 &b) {
       // Up
       nav.doNav(downCmd); // It's called downCmd because it decreases the index of an array. Visually that would mean the selector goes upwards.
    });

    btnDwn.setLongClickHandler([](Button2 &b) {
        nav.doNav(escCmd);
    });

    btnDwn.setClickHandler([](Button2 &b) {
      // Down
      nav.doNav(upCmd); // It's called upCmd because it increases the index of an array. Visually that would mean the selector goes downwards.
    });
  }
```

apronin83, 2020-04-30:
> "To solve the problem" .... something tells me that this should be a response to other issue and not an issue by itselff, am I right?

My English is very bad :)

closed 2020-04-30 by apronin83