neu-rah/ArduinoMenu #285: enter a pin code?

jeppo7712, 2020-04-16:
Is there a way to enter a pin code? For entering for example a numeric pin code one would need to be able to set each digit separately... For now I have only found a way to set a generic number with "Step size" and "fine step size" but this is not handy with for example setting a 6 digit pin code...
thanks

neu-rah, 2020-04-16:
You can use the text field and set the mask to allow only digits, however the pin will show as text.
If you want the digits to be hidden (ie: using * instead of digit) a custom draw field has to be used... that field can also be turned into a plugin.

_note: text fields allow separate character input, no range check or step/tune_

jeppo7712, 2020-04-17:
thanks, that seems to do the trick!

closed 2020-04-19 by neu-rah