neu-rah/ArduinoMenu #286: Real-time sensor display with configurable menu settings

lloydvolson, 2020-04-18:
I am sensing heart rate with real-time values on a display. A SPI SSD1351 running a display screen built with ucglib  exists in a function "monitorScreen()" . It can be called and displayed if nav.poll() is not called. Once polling begins the monitorScreen() will not co-exist with the menu system, or even take turns with it by trying to idle the menu... If I try to call monitorScreen() after polling has begun I get strange behavior.  If idleTask or exit() manages to suspend the menu and I call monitorScreen I either get a flashing monitorScreen once a second or both screens overlaid on one another. It would be nice if the EXIT (<Back) item in mainMenu would just kick me out of mainMenu instead of to the top item in mainMenior hide mainMenu so I could get back to the monitorScreen display. Nothing I have tried has worked. Some similar working example or advice would be very helpful.   Nothing in the Wiki or in "Issues", old or new, has helped.
l can send more details, code, etc.  Anyone had similar issues?

neu-rah, 2020-04-18:
It makes sense what you describe, if you  plan to use the display you should not call the menu `poll` or any menu IO functions (`doInput`/`doOutput`) as both systems will try to access the same hardware making screen flash (see https://github.com/neu-rah/ArduinoMenu/wiki/Idling#the-efficient-way). There might also be some hints on wiki FAQ.

one can:
- control timeout duration by setting nav. root `timeOut` (in seconds)
- prevent exit/timeout by setting root nav. `canExit` to false
- use timeout to start other process by calling it from the idle handler functions or by using idle handler to set a flag
- know if menu entered timeout (idle state) by checking if `nav.sleepTask` is NULL

just let me know if i can help further

lloydvolson, 2020-04-18:
Thanks Rui, for a great library and excellent support. I tried the first
two things and will see about the other two.
Right now I am going to try to implement a State Machine with a switch case.

It would be great if there were a way to get the EXIT item in the mainMenu
to disable polling and clear the screen until called
again with a button press. Can that be tweaked in nav.cpp or nav.h with
some if statement or bool?
Maybe the State Machine option will work. I'll work on it today and let you
know. Thanks again.
to the real-time monitorScreen after setting configurations.

neu-rah, 2020-04-18:
Thanks @lloydvolson
I can't disable polling that way because the menu would then idle forever, polling is responsible to put the menu out of idle state too.
Suspending polling is therefor a thing that MUST be done by user code... as well as resuming it, you will probably also need a way to signal that the menu system should redraw after you have used the screen and to reset the inactivity counter.
We could however add some functions to help users in this tasks.

lloydvolson, 2020-04-26:
Thanks.  I'm still blocked from proceeding. I have a good MainMenu a
SubMenus that work well but can not 1) start out with the real-time sensor
User Screen, go to the MainMenu to adjust settings, and then exit the
MainMenu to get back to the user screen.

The Exit (Back) menu item does not exit the MainMenu, just scrolls back to
the top item. What I would like to do is:

1) turn my device on and see my data coming in on the User Screen  --this
works well.
2) do a* longer* button-press to get to the MainMenu and adjust the
Settings like setting the clock-- I can do this by starting nav.poll()
    but if I do this I get both the MainMenu and my user screen. I can't
really get my user screen to disappear. I have tried moving it out
of loop(), I have tried to use a State Machine (still working on it). I can
scroll down but it is a mess and can't see much of the MainMenu.   The
numbers in the Test item, "55%"   are blown up large to the size of
large-font numbers I have on the User Screen, I can see cursor
high-lighting but most things
     are blacked out.

   What i want to happen is to either have the User Screen become
transparent, draw the MainMenu, go to the SubMenu Settings,
   Exit the Submenu, Exit the MainMenu.
3) Exit the Settings Menu  to go Back to Main Menu-- this works well
4) Exit the MainMenu to go back to the user screen. But the Exit (Back)
option only returns me to the top of the MainMenu.

You said "Suspend" the MainMenu. How to do that? I think I have tried Idle
Task and Timeout, maybe I'm doing it wrong.

Then "a way to signal that the menu system should redraw after you have
used the *screen?* and to reset the inactivity counter."
First, by "screen" do you mean my User Screen? I will not use the Menu
System very much at all except to turn certain
options on, off or adjust values.

"Inactivity counter". Do you mean IdleTimeout?  Can I just have the timeout
continue until there is a long button-press?

For the "Exit-Back" item in the MainMenu:  it would be perfect if a short
button-press just did a downCmd and that returned the
cursor to the top, like it is doing now.  But if I  could program a* longer*
button-press to suspend polling, return to my User Screen.

Then once back in the User Screen I could use a *longer *button-press to
clear the screen ("ucg.clearScreen();"), and maybe end
an IdleTimeout , see the MainMenu again to use it, then "Exit" when done
and return to the User Screen.

You said "polling is responsible to put the menu out of idle state". Could
the MainMenu just go black so my User Screen can be
seen? Is that "dirty" or "LCD= NULL"? There is no explanation in the wiki
or in your forum questions of how to use that.

Would it be helpful to show you a video of what I am trying to do? I have a
watch-size device (pulse oximeter) developed in China
13 years ago using an SSD 1351 OLED and one button to enter the Settings,
adjust the Settings, return to the User Screen, just
like I want to do. They were able to figure out how to do it 13 years ago.
The do have two microprocessors, one that does the sensing
 and the other that does all the display stuff and settings stuff. Not sure
if that makes any difference. If it would help I could do a video
of it working.

I think the answer will be to find a way to turn nav.poll() on and off. How
do you turn it off? I think I can restart it again in code. Is the
only way to do a TimeOut? There must be a way to just turn it off and back
on. Can you help?

Thank you,