neu-rah/ArduinoMenu #287: variable 'setLedShadowRaw' with dynamic initialization put into program memory area

ForthDude, 2020-04-19:
Hello neu-rah,

I am playing with ArduinoMenu on Arduino Uno with SSD1306 OLED display and a rotary encoder. I started out with 
 examples/SSD1306Ascii/Button_Navigation/Button_Navigation.ino
which compiles and loads fine in Ardiono 1.8.12. 

As soon as I add 
 #include <menuIO/rotaryEventIn.h>
to the code I get these error messages during compile:

OLEDTest2: In function 'global constructors keyed to 65535_0_OLEDTest2.ino.cpp.o':
OLEDTest2:101:1: error: variable 'setLedShadowRaw' with dynamic initialization put into program memory area
 TOGGLE(ledCtrl, setLed, "Led: ", doNothing, noEvent, noStyle //,doExit,enterEvent,noStyle

Am I doing anything stupid? 
Is there an incompatibility with the rotary button code? 
Do you have any advice for how I can get around this? 

Thank you,
Steffen

neu-rah, 2020-04-19:
hi!
can you try other example?
also, what board are you using?

ForthDude, 2020-04-22:
Hello neu-rah again,
I tried different examples, the only other one I could find that does this is:
examples/rotaryEvent/rotaryEvent/rotaryEvent.ino

In order to get this one to compile I had to modify line 155 
Wire.begin(SDA,SCL);
to 
Wire.begin();
to make it work with the wire library in  Arduino 1.8.12 under Linux64.

After that change it generates a similar error as I described in the original post:
variable 'fieldShadowRaw0' with dynamic initialization put into program memory area

I am using an Arduino Uno board.

If I uncomment the 
#define MENU_USERAM
in ~/Arduino/libraries/ArduinoMenu_library/src/macros.h

then this file and also the file I was talking about in the original post both compile but there is hardly any memory left for other things to do. The rotaryEvent demo won't even fit into the Arduino UNO...

So, is there something I overlook or  any better way to get around this? 

Thank you, 
Steffen

neu-rah, 2020-04-22:
Hi Steffen!

Removing `#define MENU_USERAM` will make no usage of flash memnory and probably the sketch will be tight or wont fit the RAM or an UNO.
And yes some examples simply wont fit on uno and some are ESP specific, it seems tio be the case of the sketches on error (non AVT's will by default use RAM). 
Using uno with a graphics library most of the time makes menu unusable, unless one uses an ascii librtary on TFT/Oled screen (SD1306Ascii or U8X8).
Uno will play well with text LCD's or Serial IO.

things like that can be fixed if the sketch does not use othe MCU specif features like wifi.

Thanks for bringing this to my attention.