neu-rah/ArduinoMenu #288: Use updateEvent to trigger a callback in choose or select fields

waisens, 2020-04-23:
Hi rui:
I am playing with Arduino again.
Is it possible to call a callback function when updating a value selected by a "select" or "choose" field?
>	Menu::select<byte>& Modes = *new Menu::select<byte>("mode", mode, sizeof(Modes_Data) / sizeof(prompt*), Modes_Data, (Menu::callback)mode_cb, updateEvent);

With a toggle field it works using "enterEvent".
>	toggle<byte>& Modes = *new toggle<byte>("mode", mode, sizeof(Modes_Data) / sizeof(prompt*), Modes_Data, (Menu::callback)mode_cb, enterEvent);

But with the fields "select" or "choose" I cannot use "enterEvent" and when I use "updateEvent" the function is not called.

Note: "updateEvent" works correctly when I edit numerical values for example

> menuField <typeof(deltaVal)>	&Delta		= *new menuField<typeof(deltaVal)>	(deltaVal,	"delta",	"m",  0.1,	10.0,	1,	0.1,	(Menu::callback)deltaVal_cb, updateEvent);

Thank you in advance

Esp32 core 1.01
ArduinoMenu 4.17.25 
Constructor: U8G2_SSD1306_128X64_NONAME_F_HW_I2C

neu-rah, 2020-04-23:
yes, this example might help, some events can be handed on the parent element, and there are also focus/blur events.

https://github.com/neu-rah/ArduinoMenu/blob/master/examples/handlers/handlers/handlers.ino

let it run and watch the printout

just let me know if you need more help

waisens, 2020-04-23:
Thank you!!!
"exitEvent" works as I expected

> Menu::select& Modes = new Menu::select("mode", mode, sizeof(Modes_Data) / sizeof(prompt), Modes_Data, (Menu::callback)mode_cb, exitEvent);

> Menu::choose& Modes = new Menu::choose("mode", mode, sizeof(Modes_Data) / sizeof(prompt), Modes_Data, (Menu::callback)mode_cb, exitEvent);

closed 2020-04-23 by waisens