neu-rah/ArduinoMenu #289: License compatibility issue

pat1, 2020-04-25:
I suspect it could be some problems of compatibility with your non commercial use license.
If I am not wrong there are possibility we cannot link this library with some other library with license that is not compatible with non commercial clause as for GPL license.

https://fedoraproject.org/wiki/Licensing:Main?rd=Licensing#Bad_Licenses_3
Creative Commons NonCommercial, any version (#CC-BY-NC)
"This license does not qualify as free, because there are restrictions on charging money for copies. "

As for Covid19 there are any possibility to "relax" your software license to a  Open Source License for example with one approved by Open Source Initiative  https://opensource.org/licenses ?

I am a developer and I use GPL license for some libraries and project and I want contribute and use your library.

neu-rah, 2020-04-25:
Thanks for enlighten me.
would LGPLv2+ be ok?

pat1, 2020-04-25:
Thank you for making yourself available.
 LGPLv3+ should be ok.
If you want more compatibilities LGPLv2.1+ is the more compatible.
avr arduino core and esp8266 arduino core for example use GNU Lesser General Public License v2.1
https://github.com/esp8266/Arduino/blob/master/LICENSE

License compatibility is like a scienze:
https://fedoraproject.org/wiki/Licensing:Main?rd=Licensing#GPL_Compatibility_Matrix

neu-rah, 2020-04-26:
done it, LGPL-2.1+

pat1, 2020-05-05:
perfect!

closed 2020-05-05 by pat1