neu-rah/ArduinoMenu #290: Compatability with Adafruit 1.8" TFT V2 ?

PeteMoog, 2020-04-28:
hi 
I am using a a mega 2560 with the Adafruit 1.8" TFT + joystick and 3 button shield (it's the V2 Shield) here https://www.adafruit.com/product/802 

Is it possible to use ArduinoMenu with this shield? the shield uses i2c seesaw chip to communicate and I am struggling to get anything running (or even to compile) to use this board and ArduinoMeno, If there is any sample code that someone could direct me to it would be very helpful. even some ideas to the most suitable libraries or includes would be very helpful.

Do I use 

#include <Adafruit_GFX.h>
#include <Adafruit_ST7735.h>

from the shield example... OR

#include <TFT.h> 

that  rui uses in his example "menu_tft_test.ino"

This is my Arduio and first project, I fear I have bitten off more than I can chew!

FYI the board and shield are wired up correctly and the following example runs fine... https://learn.adafruit.com/1-8-tft-display/testing-the-shield

thanks in advance....

Pete

neu-rah, 2020-04-28:
Hi Pete!

It should work...

- pick an adafruit example that works for your screen
- then grab the adafruit gfx object construction and transport to the adafruit menu example
- remove the menu example existing display device construction but take note of its name
- then replace its name by the name of your transported object... or rename your transported object to match the existing

dell free to add error messages or code to this issue :+1: 

rui

PeteMoog, 2020-04-28:
many thanks for the quick reply, i am on child care duties in the morning but will try as soon as i can get to my desk, 👍

PeteMoog, 2020-05-23:
Hi again
I have been working on this frequently since we last spoke but have been reluctant to ask for help as my C++ and arduino knowledge is so basic I thought I had better RTFM first, which I now have, my understanding is slightly greater but much of it is still a mystery.

I now have an example compiling and the screen displaying the default menu, I also know the buttons on the shield are being read via the SPI as I have put some code in the Void loop() to read the buttons using the shield's seesaw/SPI and when I press a button or move the joystick I see these being printed in the serial monitor view (i am using eclipse/sloeber).

`#include <Arduino.h>

/********************
Sept. 2014 ~ Oct 2016 Rui Azevedo - ruihfazevedo(@rrob@)gmail.com

menu with adafruit GFX
output: 1.8" TFT 128*160 (ST7735 HW SPI)
input: Serial + encoder
www.r-site.net

alternative encoder (clickEncoder) uses:
https://github.com/0xPIT/encoder
https://github.com/PaulStoffregen/TimerOne
***/

//removed as extra libs are not included into travis
// #define USE_CLICK_ENCODER

#include <SPI.h>
#include <Adafruit_GFX.h>
#include <Adafruit_ST7735.h>
#include <menu.h>
#include <menuIO/adafruitGfxOut.h>
#ifdef USE_CLICK_ENCODER
  #include <TimerOne.h>
  #include <ClickEncoder.h>
  #include <menuIO/clickEncoderIn.h>
#else
  #include <menuIO/encoderIn.h>
#endif
#include <menuIO/keyIn.h>
#include <menuIO/chainStream.h>
#include <menuIO/serialOut.h>
#include <menuIO/serialIn.h>
#include <Adafruit_seesaw.h>
#include <Adafruit_TFTShield18.h>

Adafruit_TFTShield18 ss;

using namespace Menu;

//ST7735 1.8TFT 128x160 This was original by RUI - replaced below by example from seesaw_shield18_test
// #define TFT_CS   A1
// #define TFT_DC   A0
// #define TFT_RST  A2

// #define SD_CS    4  // Chip select line for SD card on Shield			//from seesaw_shield18_test
#define TFT_CS  10  // Chip select line for TFT display on Shield		//from seesaw_shield18_test
#define TFT_DC   8  // Data/command line for TFT on Shield				//from seesaw_shield18_test
#define TFT_RST  -1  // Reset line for TFT is handled by seesaw!		//from seesaw_shield18_test

Adafruit_ST7735 gfx(TFT_CS, TFT_DC, TFT_RST);

#define LEDPIN A3

// rotary encoder pins
#define encA    2
#define encB    3
#define encBtn  4

result doAlert(eventMask e, prompt &item);

int test=55;

int ledCtrl=LOW;
[... 199 lines cut ...]
	    buttonhistory |= 64;
	  }
	  if (! (buttons & TFTSHIELD_BUTTON_IN)) {		// Button IN
		  buttonhistory |= 128;
	  }
	  if (oldbuttonhistory != buttonhistory) {
		  serial.println(buttonhistory);
	  }
	  oldbuttonhistory = buttonhistory;
//	  Serial.println(TFTSHIELD_BUTTON_1);
//	  Serial.println(buttons);
  nav.poll();//this device only draws when needed
  digitalWrite(LEDPIN, ledCtrl);
  delay(1000);//simulate a delay when other tasks are done
  buttonhistory = 0;
}
`

What I am completely stuck on is how to I transfer the button presses on the shield into Serial input to control ArduinoMenu?

Any example code or examples would be greatly appreciated. 

Thanks in advance

P

PeteMoog, 2020-05-26:
Hi Rui
I have found an example called  "codeCtrl.ino" and noticed the "nav.donav(upCmd)" statements, I added these to my loop section as follows and I now have a working screen and the joystick almost works. I cant seem to enter any submenus, or change the 55% in the example, why could that be?

As you can see I started trying to write commands to serialIn so I could concatenate this stream using Chainstream and subsequently after a bit more reading I would have used MENU_INPUTS after realising this does the same thing as chainstream. Would any of the previous methods I mention been better that method I am posting now? 

I want to use ArduinoMenu to control a few stepper motors, they must run uninterrupted and run at a defined speed and defined number of steps, I will use the menu so I can set these parameters. The duration of this may be as much as 30 minutes and the steppers should run uninterrupted with the exception of a possible ABORT command driven from a dedicated hardware button on the shield should things go wrong in the physical world. 
```
void loop() {

// Following block reads switch inputs from SPI interface (seesaw) copied from seesaw_shield18
	uint32_t buttons = ss.readButtons();
	  if(! (buttons & TFTSHIELD_BUTTON_DOWN)){		// DOWN
		  buttonhistory |= 1;
//		  strIn.write(downCmd);
		  nav.doNav(downCmd);
	  }
	  if(! (buttons & TFTSHIELD_BUTTON_LEFT)){		// LEFT
	    buttonhistory |= 2;
//	    strIn.write(leftCmd);
	    nav.doNav(leftCmd);
	  }
	  if(! (buttons & TFTSHIELD_BUTTON_UP)){		// UP
	    buttonhistory |= 4;
//	    strIn.write(upCmd);
	    nav.doNav(upCmd);
	  }
	  if(! (buttons & TFTSHIELD_BUTTON_RIGHT)){		// RIGHT
	    buttonhistory |= 8;
//	    strIn.write(rightCmd);
	    nav.doNav(rightCmd);
	  }
	  if(! (buttons & TFTSHIELD_BUTTON_1)){			// Button 1
	    buttonhistory |= 16;
//	    strIn.write(Button1=1);
	  }
	  if(! (buttons & TFTSHIELD_BUTTON_2)){			// Button 2
	    buttonhistory |= 32;
//	    strIn.write(Button2=1);
	  }
	  if(! (buttons & TFTSHIELD_BUTTON_3)){			// Button 3
	    buttonhistory |= 64;
//	    strIn.write(Button3=1);
	  }
	  if (! (buttons & TFTSHIELD_BUTTON_IN)) {		// Button IN
		  buttonhistory |= 128;
//		  strIn.write(idxCmd);
		  nav.doNav(upCmd);
	  }
	  if (oldbuttonhistory != buttonhistory) {
		  serial.println(buttonhistory);

	  }
	  oldbuttonhistory = buttonhistory;
//	  Serial.println(TFTSHIELD_BUTTON_1);
//	  Serial.println(buttons);
  nav.poll();//this device only draws when needed
  digitalWrite(LEDPIN, ledCtrl);
  delay(1000);//simulate a delay when other tasks are done
  buttonhistory = 0;
}
```

PeteMoog, 2020-05-26:
PS. I also joined the Gitter bbut I think people are away on bank holidays...

neu-rah, 2020-05-26:
Hi Pete

Thanks for sharing the code...
yes, gitter was a bit abandoned but text stays there, I'm on a kindo of holiday cabling my home made CNC.

PeteMoog, 2020-05-26:
Stay on holiday, please dont talk to me again until it's Finished 👍 

When you are back from holiday tell me how big your CNC is.

PS. fixed the select menu thing, it was a typo.

PeteMoog, 2020-05-30:
All almost work, as few small glitches with secondary panel having blank lines when scrolling down the main panel but sure its something I have done. Will close this issue and open a fresh one if I can resolve it. Thanks for your help Rui.

closed 2020-05-30 by PeteMoog