neu-rah/ArduinoMenu #297: How to go to sub-menu using code ( not considering the current position )

GoDotMad, 2020-05-15:
I want to move to a certain sub-menu whenever something happens regardless of the current position or level of the menu.

I think I should use the navRoot to check witch level it is and then set the target index and then do a entercmd  but I'm not sure .

secondly , can I display more than one value with a label in a single line?

neu-rah, 2020-05-15:
yes and yes

you can poke with the navRoot indeed, however there are also commands to move, and select by index, see the codeCtrl example here https://github.com/neu-rah/ArduinoMenu/blob/master/examples/codeCtrl/codeCtrl/codeCtrl.ino

if compiled with MENU_ASYNC defines at compiler command line the you have also **async API**

with some extra commands

**escTo(int level)** move back to level (if on higher level), useful to move to root `escTo(0)` . This will traverse all levels and emit respective events as if user escaping toll the desired level is reached.

**async(const char\* at)** accepts a path of zero based indexes ie: "/0/1/0" and navigate to desired node, a transition from current state to the desired node is respected in terms of events.
Can also add a final value to set field value or enumerated index, Async mode is default on esp8266 because of web interface.

You can combine multiple menu elements (usually fields) in a single line (PAD menu). Most examples include so as IP fields, or date/time fields, that are in reality just a set of fields.

hope it helps

GoDotMad, 2020-05-15:
Thanks 
I don't know why I didn't notice that when I was reading the " navigation " section of the wiki

GoDotMad, 2020-05-16:
is there a reason why I can't do two .doNav() commands after each other?

GoDotMad, 2020-05-16:
it turned out that I can't do enterCmd after idxCmd 

any thing else is fine

GoDotMad, 2020-05-16:
idxCmd does enterCmd before and after arriving to the selected index

is that how it is supposed to work?

closed 2020-05-16 by GoDotMad

neu-rah, 2020-05-16:
https://gitter.im/ArduinoMenu/Lobby