neu-rah/ArduinoMenu #298: BUG, MAX_DEPTH, Example targetSel, LCD 16x2 I2C

k3v1n97, 2020-05-16:
Hello Sir @neu-rah 

there is a bug in the code, if I initialize max_depth = 2 on LCD 1602, targetEdit doesn't appear, instead it restarts

if I initialize max_depth = 4 on the LCD, targetEdit appears, but not all TextField (<name>) are listed on the LCD

Please fix the bug, I really need it for my project

Thanks

neu-rah, 2020-05-16:
Hi!

you have to setup MAX_DEPTH according to your menu structure needs. Set it to the max depth you can can on any branch, being that each sub-menu counts as one, and also esch enumerated fiels (SELECT, CHOOSE, TOGGLE).

make sure you set the `dataSz` to match the data size on your data array.

neu-rah, 2020-05-17:
Indeed there were multiple problem with that example and a scroll bug.

Fixed the scroll bug (just pushed) and on the code I've done a couple of changes. 

- First reduced the array size to 4 because I was using debug and memory (RAM) usage wa already above 90% on Uno.
- Second replaced that `strcpy` by `strncpy(name,o.name,nameSz);` that copy was messing with remaining data and overriding.

Text edit fields use a dimensioned space and initialize the size by the initial string data, therefor the extra spaces on original data.

Still the zero termination (on the original was overridden), still need to check that.

The data override was causing the second fields validation to panic with invalid range. Field variable are required to be in range just because we have no default to reset them if they are not.

Please let me know how it went on your display, i was using LiquidCrystal_PCF8574 on my side.

thanks for testing reporting this.

k3v1n97, 2020-05-17:
> Indeed there were multiple problem with that example and a scroll bug.
>
> Fixed the scroll bug (just pushed) and on the code I've done a couple of changes.
>
> * First reduced the array size to 4 because [...]

it works, according to my expectations.

Thanks for reply and fix the bug :)

closed 2020-05-17 by k3v1n97