neu-rah/ArduinoMenu #300: the size of menu and submenu locations

nik-nikolaev, 2020-05-16:
Hello ...
I am interested in a couple of questions: are there any restrictions on the number of menu items? and the second question: in this implementation on the screen, I have the main menu in the first column, and the second-level menu (submenu) opens in the second column. is there such a configuration (maybe I missed something) where the submenu opens in the first column, and is it possible for the third-level menu to be drawn in the second column accordingly?

```
#define MAX_DEPTH 4
#define textScale 2
MENU_OUTPUTS(out, MAX_DEPTH
             , ADAGFX_OUT(tft, colors, 6 * textScale, 9 * textScale, {0, 0, 14, 12}, {14, 0, 14, 12})
             , SERIAL_OUT(Serial)
            );
```

nik-nikolaev, 2020-05-16:
[here](https://github.com/neu-rah/ArduinoMenu/issues/299) here, described something similar, I would like to hear your vision of the issue.

neu-rah, 2020-05-16:
Hi!
yes, there is a limit on the number of items, they can be 127 on AVR mcu's and 32767 on non AVR's.

about the columns, they are called `panels` on the code and wiki, and you can specify multiple..
please reduce that part to a single column covering all screen
`{0, 0, 14, 12}, {14, 0, 14, 12}` -> `{0,0,28,12}`

panels allow previews and submenu open on extra panels if available, there is also a config to reserve one panel for previews, this is a thing I'm not yet sure to support on AM5 because usage has been very rare