neu-rah/ArduinoMenu #301: HOW TO ADD A CAPTION OR A VALUE TO A MENU OR SUB-MENU HEADER?

NeftaliLoya, 2020-05-16:
I need to add a variable's value on the right side of a menu or a sub-menu header. Please, see the following example:
MENU(mainMenu,"*** MAIN MENU ***",doNothing,noEvent,wrapStyle
  ,SUBMENU(SubMenu 1) **HOW TO ADD SubMenu 1's VALUE OR A TEXT STRING HERE?**
  ,SUBMENU(Sub-Menu 2) **HOW TO ADD SubMenu 2's VALUE OR A TEXT STRING HERE?**
  ,SUBMENU(Sub-Menu 3) **HOW TO ADD SubMenu 3's VALUE OR A TEXT STRING HERE?**
  ,EXIT("Save changes and exit Main Menu")
);

My need goes as follows:
1.- My Main Menu has three sub-menus and en exit. Evidently, each of those sub-menus is linked or tied to a variable that changes its value according to a chosen choice inside of it.
2.- Assuming that Sub-Menu 1 holds 3 fields to choose and I select field 2 as the desired choice. Internally, the macro assigns a value of 100 to the associated Sub-Menu 1's variable.
3.- As I leave Sub-Menu 1, I do not have a way of telling which value I selected for SubMenu 1 and so it goes for SubMenu 2 and SubMenu 3.
4.- Is there a way of adding a text string on the right side of those SubMenus to keep track of their assigned chosen values?

English is not my first language and often times I lack the necessary eloquence to make myself understandable.

Joennuh, 2020-05-16:
I'm not sure but isn't it a CHOOSE you need instead of a submenu?

NeftaliLoya, 2020-05-16:
I am using a choose macro as the sub-menu. I fell a victim to my own orderliness! As I aligned the text strings so they show up neatly with the menu's structure, the additional spaces that I used got in the way. The caption was there; however, it was filled with blank spaces! LOL!!!! Please, forgive me! I promise to make my best effort before asking for help! Thanks for your undeserved support!

closed 2020-05-16 by NeftaliLoya