neu-rah/ArduinoMenu #303: Hello...

NeftaliLoya, 2020-05-21:


NeftaliLoya, 2020-05-21:
Hello everyone…
The menu enters into the idle mode when the encoder's selection button is pressed for a long time. This peculiarity renders its use inoperative.

My application triggers the navigation mode by pressing the encoder button for three seconds or more; however, having pressed the encoder during all that time, results in the menu going into idle mode. That is, it causes the navigation mode to be canceled immediately after entering it through having pressed the encoder button for a long time.

The application I'm using is 100% based on the adafruitGfx / tft / tft.ino example code that comes in the package that I downloaded from github.

I will appreciate any help you could provide me to resolve this conflict.

Best regards,

neu-rah, 2020-05-21:
Hi!
I would handle the encoder input when menu in idle state or all the time to have more control over it, see the codeCtrl example, you can issue navigation commands and use any input method.

if you provide the input initialization lines i can be more specific

NeftaliLoya, 2020-05-21:
Hello Rui…
I am using your adafruitGfx example as a template. The input initialization lines are the same…

   encoderIn<encA,encB> encoder;
  encoderInStream<encA,encB> encStream(encoder,1);
  keyMap encBtn_map[]={{-encBtn,defaultNavCodes[enterCmd].ch}};
  keyIn<1> encButton(encBtn_map);
  MENU_INPUTS(in,&encStream,&encButton);

It still send me to the idle mode when I keep the encoder's button pressed for 1.5 secos or more!
Any suggestions?

Neftali Loya Garibay
neftali.loya@icloud.com

neu-rah, 2020-05-22:
Hi!
This is specific to key/button handling used with the builtin encoder driver. It considers a long press as an escape.

you can define `ESCAPE_TIME` at compiler command line to be sufficient high. That alone might not be sufficient because the menu will be controlling inactivity time and might enter idle just by that, once you have external control of clicking you should also call `idleOff` when resuming the menu to ensure it does not enter idle by timeout.

NeftaliLoya, 2020-05-22:
Good morning Mr. Azevedo. Thank you for your infinite kindness and your cordial diligence in answering and clarifying all these doubts. Saying thank you is saying little compared to what you and your work inspire us.

Turning to the matter at hand, I am pleased to tell you that I changed the detection time of a long press to 99.99 seconds just to be safe. Now, I can enter the menu by pressing the encoder for 4 seconds and finally discovered the problem.

When I enter the menu for the first time, the focus of the cursor is on option one of the main menu and, to exit, obviously I must select the option “EXIT”. When I enter the menu again, the focus of the cursor remains recorded in the “EXIT” option, so when releasing the encoder button, the EXIT option is automatically selected.

This would be resolved if it were possible to enter the menu by preconfiguring the initial cursor position. Would it be possible to do that? That is, can the cursor position be preconfigured before entering navigation mode?

Best regards,
Naftalí Loya Garibay

NeftaliLoya, 2020-05-23:
Good evening! Problem solved... other files were involved in the ESCAPE_TIME issue. softKeyIn.h, KeyIn.h, and AltKeyIn.h. I set 99.99 seconds as the required time to acknowledge a long-press in all of them all and now I can control when and how to enter and leave the navigation mode. The issue with setting the cursor's focus on the first item from the main menu was solved by calling the nav.reset() function at the time of leaving the navigation mode. Thanks!

closed 2020-05-23 by NeftaliLoya

neu-rah, 2020-05-23:
still, even not having the exit option on focus, it could be another equally "dangerous" function like missile launch ;)
The issue there was with button release, you should wait for button release before delivering control to the menu because that what activates the action. But guess you already figured that :D