neu-rah/ArduinoMenu #307: Issue with SAMD21G18 boards

pipesg, 2020-05-29:
Hi,
Firstly, many thanks for your great work developing and maintaining this useful library. I used this library in a project to control a device by sending command using LoRa. The first prototype used Feather 32u4 and worked fine, now I designed a pcb and instead Mega32u4 I have used a SAMD21G18 to increase the memory available for the application. When I compiled the application using Feather M0 board I got this warning during compilation 

> invalid conversion from 'const Menu::panel*' to 'Menu::panel*' [-fpermissive]

I tried again using Feather 32u4 and it compile fine.
I also tried to compile your example SSD1306Ascii-Button_Navigation, in which I based my application, with Feather M0 and MKRZERO boards and I got the same warning, this is the full output when compiling:

Button_Navigation:152:`` error: invalid conversion from 'const Menu::panel*' to 'Menu::panel*' [-fpermissive]
panelsList pList(panels, nodes, 1); //a list of panels and nodes
In file included from D:\GPS_Logger_new\Arduino\libraries\ArduinoMenu_library\src/menuIo.h:178:0,
from D:\GPS_Logger_new\Arduino\libraries\ArduinoMenu_library\src/menuDefs.h:62,
from D:\GPS_Logger_new\Arduino\libraries\ArduinoMenu_library\src/menu.h:15,
from D:\GPS_Logger_new\Arduino\libraries\ArduinoMenu_library\examples\SSD1306Ascii\Button_Navigation\Button_Navigation.ino:25:
D:\GPS_Logger_new\Arduino\libraries\ArduinoMenu_library\src/nav.h:22:16: note:   initializing argument 1 of 'Menu::panelsList::panelsList(Menu::panel*, Menu::navNode**, Menu::idx_t)'
inline panelsList(constMEM panel p[],navNode* nodes[],idx_t sz):panels(p),nodes(nodes),sz(sz) {
exit status 1
invalid conversion from 'const Menu::panel*' to 'Menu::panel*' [-fpermissive]

Any help to fix this error will be very welcome.
Thanks in advance!

neu-rah, 2020-05-29:
Hi! Thanks for reporting this.

I was not able to reproduce it using platformio or arduino IDE, and some boards MKRZERO and Feather among others.

This parameter should alway be accepted `const` as panels are never changed by the menu system however the parameter is defined on the library to be `const` on AVR's and non `const` on other platforms (as flash), should not be this way, however before changing it I would like first to reproduce it.

To quick solve your problem you should have a `constMEM panel panels[] MEMMODE = ...` above that line, please remove the `constMem` from that, does it solve?

if you can share more details, ie: what IDE are you using or any other pertinent info I will appreciate.

It seems that sketch and library are being build with different defines, are you defining any of `USE_PGM`, `MENU_USERAM`, `MEMMODE`, `USING_PGM`, `USING_RAM` in your sketch?

pipesg, 2020-05-29:
Hi,
Obrigado!
I am using arduino IDE 1.8.5 and I also tried with visual studio 2019 with visual micro. I am using the example SSD1306Ascii-Button_Navigation without any more definitions . Sorry I attached the output from my program not from the example, anyway no defines added. These are the warnings of the SSD1306Ascii-Button_Navigation:

Button_Navigation:152:` error: invalid conversion from 'const Menu::panel*' to 'Menu::panel*' [-fpermissive]
 panelsList pList(panels, nodes, 1); //a list of panels and nodes
In file included from D:\GPS_Logger_new\Arduino\libraries\ArduinoMenu_library\src/menuIo.h:178:0,
from D:\GPS_Logger_new\Arduino\libraries\ArduinoMenu_library\src/menuDefs.h:62,
from D:\GPS_Logger_new\Arduino\libraries\ArduinoMenu_library\src/menu.h:15,
from D:\Archivos temporales\TMP\arduino_modified_sketch_852239\Button_Navigation.ino:25:
D:\GPS_Logger_new\Arduino\libraries\ArduinoMenu_library\src/nav.h:22:16: note:   initializing argument 1 of 'Menu::panelsList::panelsList(Menu::panel*, Menu::navNode**, Menu::idx_t)'
inline panelsList(constMEM panel p[],navNode* nodes[],idx_t sz):panels(p),nodes(nodes),sz(sz) {
exit status 1
invalid conversion from 'const Menu::panel*' to 'Menu::panel*' [-fpermissive]

Before I saw your mail, trying some other examples of your library I observed that SSD1306Ascii compile without any issue in these boards (feather M0 and MKRZERO) and the only difference is that it have the constMEM (line 158) so I added to the SSD1306Ascii-Button_Navigation (line 150) and it worked fine, so apparently the issue is there, but I have still not tested the program in the board.

closed 2020-06-09 by pipesg