neu-rah/ArduinoMenu #310: Warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]

fmatray, 2020-06-03:
Hello,
During compilation, I get 23 Warnings:
```
dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing] 
```

example with this line in macros.h:280 :
```
constMEM Menu::promptShadow& opShadow##cnt=*(promptShadow*)&opShadowRaw##cnt;\
```

I try to find why this happen but I have no clue.

neu-rah, 2020-06-03:
long story short: this has to do with static allocating objects on flash and virtual tables. C++ won't support this so i did this shadow allocation on flash and complicated cast to have part of the object on flash... not a good idea, but it was the best I knew some year ago.
I'm removing the warning now by using an union, it seems to work but will require some more testing.
Not doing so on AM5, and still gaining flexibility because we have no longer to opt by flash or ram allocation and still can mix them.

fmatray, 2020-06-03:
Sounds great if you have a solution. I have 25 remaining warnings during compilation with the menu and all are related to this.

Is you need some more testing, I can on a mkr 1010 and LCD1602.

neu-rah, 2020-06-03:
I'm hitting them, some about char to const char and some about using string literal... almost done (I hope)

neu-rah, 2020-06-03:
done, also done an extensive review on the examples.
hope i'm not creating new bugs with this, please report any issue
thanks

fmatray, 2020-06-04:
I test this morning with your commits. 
I have no more warning and the menu works on a MKR 1010.
I'll report if I find any strange behavior or bug.

Thanks a lot.

closed 2020-06-04 by fmatray