neu-rah/ArduinoMenu #312: Number of menu items.

Astar75256, 2020-06-05:
Hello...
Please tell me how you can increase the number of menu items for a macro. Maximum managed to make only 15 elements.

MENU(funcColorMenu1, "Functions 2", doNothing, noEvent, wrapStyle,
	OP(RANDOM_COLOR_EFF, action2, anyEvent),
	OP(SMOOTH_RANDOM_EFF, action2, anyEvent),
	OP(PRIDE_EFF, action2, anyEvent),
	OP(RED_STROBE_EFF, action2, anyEvent),
	OP(PUPRLE_STROBE_EFF, action2, anyEvent),
	OP(YELLOW_STROBE_EFF, action2, anyEvent),
	OP(GREEN_STROBE_EFF, action2, anyEvent),
	OP(CYAN_STROBE_EFF, action2, anyEvent),
	OP(BLUE_STROBE_EFF, action2, anyEvent),
	OP(ALL_COLORS_STROBE_EFF, action2, anyEvent),
	OP(MULTICOLORS_STROBE_EFF, action2, anyEvent),
	OP(RED_FADE_EFF, action2, anyEvent),
	OP(YELLOW_FADE_EFF, action2, anyEvent),
	OP(GREEN_FADE_EFF, action2, anyEvent),
	// OP(CYAN_FADE_EFF, action2, anyEvent),
	// OP(BLUE_FADE_EFF, action2, anyEvent),
	OP(PURPLE_FADE_EFF, action2, anyEvent),
	EXIT("<Back")
);

Only two elements were missing, so I commented them out)

Astar75256, 2020-06-05:
`MENU(funcColorMenu1, "Functions 2", doNothing, noEvent, wrapStyle,
	OP(RANDOM_COLOR_EFF, action2, anyEvent),
	OP(SMOOTH_RANDOM_EFF, action2, anyEvent),
	OP(PRIDE_EFF, action2, anyEvent),
	OP(RED_STROBE_EFF, action2, anyEvent),
	OP(PUPRLE_STROBE_EFF, action2, anyEvent),
	OP(YELLOW_STROBE_EFF, action2, anyEvent),
	OP(GREEN_STROBE_EFF, action2, anyEvent),
	OP(CYAN_STROBE_EFF, action2, anyEvent),
	OP(BLUE_STROBE_EFF, action2, anyEvent),
	OP(ALL_COLORS_STROBE_EFF, action2, anyEvent),
	OP(MULTICOLORS_STROBE_EFF, action2, anyEvent),
	OP(RED_FADE_EFF, action2, anyEvent),
	OP(YELLOW_FADE_EFF, action2, anyEvent),
	OP(GREEN_FADE_EFF, action2, anyEvent),
	// OP(CYAN_FADE_EFF, action2, anyEvent),
	// OP(BLUE_FADE_EFF, action2, anyEvent),
	OP(PURPLE_FADE_EFF, action2, anyEvent),
	EXIT("<Back")
);
`

neu-rah, 2020-06-05:
hi!
yes, macro has that limitation... and only editing the horrible macro to overcome the limit.
But constructing without macro is an optio, what mcu are you using?

fmatray, 2020-06-05:
You can also split into several sub menu
Random -> COLOR or smooth
Strobe -> RED, PURPLE, etc.
Fade -> Yellow, etc. 

For a user, a well structured menu is more efficient than a large flat menu.