neu-rah/ArduinoMenu #313: Improvement : InfoField

fmatray, 2020-06-06:
Hello,
Is it possible to add an InfoField ?

This field is just to display a title and a value without any edition. 
This can be usefull to show some information like "Your IP:192.168.0.13" or "API key: FCUGUOZBDQSB".

For the moment I do something with like this but it's not elegant and generic.

```
result show_ip() {
  [display on the LCD]
  return proceed;
}
OP("Show IP", show_ip, enterEvent)
```

INFO(Value, "Title", Menu::doNothing, Menu::noEvent, Menu::noStyle).

neu-rah, 2020-06-06:
yes, you can set a field read only, the field can also be hooked up to a live variable and display ie: an analog read with self/auto update

fmatray, 2020-06-06:
Do you have an example, please ? 

I tried like this:
```
const char *constMEM empty[] MEMMODE = {};
EDIT("IP:", "192.168.0.1", empty, Menu::doNothing, Menu::noEvent, Menu::noStyle)
```
The is shown but when I click on this item, I enter in an edition mode without any possibility to change. It's a kind of confusing workaround as the user believes changing the value is possible. 

Moreover, an EDITField not editable is a bit confusing :-)

neu-rah, 2020-06-07:
Enable/Disable is a runtime setting, all menu item have it
example:
```c++
mainMenu[0].disable();
mainMenu[1].enable();
```

source: https://github.com/neu-rah/ArduinoMenu/blob/master/src/items.h#L47

there is still space for improvement, we don't have a function to check enabled/disabled state,
there is `enabled` member with `status` enum with values: `{disabledStatus=0,enabledStatus=1};`

src: https://github.com/neu-rah/ArduinoMenu/blob/master/src/menuBase.h#L149

but this can be confusingand hard to remember, as `enabled` name is already used we can have a `disabled()` function to returnm a boolean instead of that enum.

ie:
```c++
inline bool disabled() const {return enabled==disabledStatus;}
```

fmatray, 2020-06-07:
Thanks a lot. Works fine like this :

```
const char *constMEM empty[] MEMMODE = {};

MENU(advanced_menu, "Advanced", Menu::doNothing, Menu::noEvent, Menu::noStyle, 
[...]
EDIT("@", wificonnexion.localIP, empty, Menu::doNothing, Menu::noEvent, Menu::noStyle),
[...]
advanced_menu[3].disable();
```