neu-rah/ArduinoMenu #315: esp32 web menu not-work

LeandroLimaPRO, 2020-06-15:
will there be compatibility of the "MENU" websocket with esp32? I tried to perform a migration with ESP32 but without success.

neu-rah, 2020-06-15:
yes, its the idea, but not tested it yet
errors or libraries missing?

LeandroLimaPRO, 2020-06-15:
I'm not a websocket ace, I'm quite a beginner to tell the truth, I made some changes:
- Using WIFIMANAGER, instead of entering wifi setup all the time.
- removed any mention of "wifi.begin" or "ssid" and "password" parameters

**CODE ARDUINO**
```cpp
/********************
Arduino generic menu system
XmlServer menu example
based on WebServer:
  https://github.com/esp8266/Arduino/tree/master/libraries/ESP8266WebServer
  https://github.com/Links2004/arduinoWebSockets

Dec. 2016 Rui Azevedo - ruihfazevedo(@rrob@)gmail.com

menu on web browser served by esp8266 device
output: ESP8266WebServer -> Web browser
input: ESP8266WebSocket <- Web browser
format: xml, json

IMPORTANT!:
this requires the data folder to be stored on esp8266 spiff
Extra libraries should be present

arduinoWebSockets - https://github.com/Links2004/arduinoWebSockets
ESP8266WiFi - https://github.com/esp8266/Arduino/tree/master/libraries/ESP8266WiFi
ESP8266WebServer - https://github.com/esp8266/Arduino/tree/master/libraries/ESP8266WebServer

for development purposes some files are left external,
therefor requiring an external webserver to provide them (just for dev purposes)
i'm using nodejs http-server (https://www.npmjs.com/package/http-server)
to static serve content from the data folder. This allows me to quick change
the files without having to upload them to SPIFFS
also gateway ssid and password are stored on this code (bellow),
so don't forget to change it.

*/
#include <WiFiManager.h> // https://github.com/tzapu/WiFiManager

#include <menu.h>
#include <menuIO/esp8266Out.h>
#include <menuIO/xmlFmt.h>//to write a menu has html page
#include <menuIO/serialIn.h>
#include <menuIO/xmlFmt.h>//to write a menu has xml page
#include <menuIO/jsonFmt.h>//to write a menu has xml page
#ifndef ARDUINO_STREAMING
  #include <streamFlow.h>//https://github.com/neu-rah/streamFlow
#else
  #include <Streaming.h>//https://github.com/scottdky/Streaming
#endif
//#include <menuIO/jsFmt.h>//to send javascript thru web socket (live update)
#include <SPIFFS.h>
#include <Hash.h>

using namespace Menu;

#ifdef WEB_DEBUG
  // on debug mode I put aux files on external server to allow changes without SPIFF update
  // on this mode the browser MUST be instructed to accept cross domain files
  String xslt("http://neurux:8080/");
#else
  String xslt("");
#endif

menuOut& operator<<(menuOut& o,unsigned long int i) {
  o.print(i);
  return o;
}
menuOut& operator<<(menuOut& o,endlObj) {
  o.println();
[... 889 lines cut ...]

WebMenu:476:3: error: 'webSocket' was not declared in this scope

   webSocket.loop();

   ^

WebMenu:477:3: error: 'server' was not declared in this scope

   server.handleClient();

   ^

Foram encontradas múltiplas bibliotecas para "WiFi.h"
Usado: C:\Users\leo_l\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\WiFi
Não usado: C:\Program Files (x86)\Arduino\libraries\WiFi
exit status 1
'WStype_t' has not been declared

Este relatório teria mais informações com
"Mostrar a saida detalhada durante a compilação"
opção pode ser ativada em "Arquivo -> Preferências"

```
PS: I realized that you are Portuguese, can we communicate in Portuguese? My english is terrible

neu-rah, 2020-06-15:
Thanks for the feedback.
I gave a quick try with the original esp8266 example.. and there is a lot to change it seems, however wifimanager "ESP8266 WiFi Connection manager..." is also an esp8266 thing, so we first have to make menu work as is before going there

LeandroLimaPRO, 2020-06-15:
Thanks for the quick return. The WifiManager lib being used is specific to ESP32. It is only replacing the manual entry of wifi "begin" data. I saw no interference in it. I look forward to returning to ESP32

Koxx3, 2021-02-11:
+1000 ;)

nagarjunredla, 2021-12-26:
PR with ESP32 support (and updated example) #374