neu-rah/ArduinoMenu #316: Support for Adafruit S7789 breakout board using teensy NOT esp

fishercat503, 2020-06-17:
I want to use my ST7789 display. If I include all the headers and setup code for this display it turns on. I can't change the background.

neu-rah, 2020-06-18:
Hi!
have you tried to mess with the colors table?

fishercat503, 2020-06-24:
I have been busy moving to platformio from the Arduino IDE. I am also making some progress. I am using your btnNav sketch and trying to send the output to an ST7789.

The code is below.  It fails on line 201 with the macro
,ADAGFX_OUT(gfx,colors,6*textScale,9*textScale,{0,0,14,8},{14,0,14,8})

no instance of constructor "Menu::adaGfxOut::adaGfxOut" matches the argument list -- argument types are: (Adafruit_ST7789, const Menu::colorDef<uint16_t> [5], Menu::idx_t [4], Menu::panelsList, int, int)

I can's figure out the last four arguments to the macro.  The line I am using is from your code somewhere else in the library. I got the gfx and colors arguments to work.

Thanks for your attention. 

/**************************************************************************
  This is a library for several Adafruit displays based on ST77* drivers.

  Works with the Adafruit 1.8" gfx Breakout w/SD card
    ----> http://www.adafruit.com/products/358
  The 1.8" gfx shield
    ----> https://www.adafruit.com/product/802
  The 1.44" gfx breakout
    ----> https://www.adafruit.com/product/2088
  The 1.14" gfx breakout
  ----> https://www.adafruit.com/product/4383
  The 1.3" gfx breakout
  ----> https://www.adafruit.com/product/4313
  The 1.54" gfx breakout
    ----> https://www.adafruit.com/product/3787
  The 2.0" gfx breakout
    ----> https://www.adafruit.com/product/4311
  as well as Adafruit raw 1.8" gfx display
    ----> http://www.adafruit.com/products/618

  Check out the links above for our tutorials and wiring diagrams.
  These displays use SPI to communicate, 4 or 5 pins are required to
  interface (RST is optional).

 
  Written by Limor Fried/Ladyada for Adafruit Industries.
  MIT license, all text above must be included in any redistribution
  Adafruit invests time and resources providing this open source code,
  please support Adafruit and open-source hardware by purchasing
  products from Adafruit!
**************************************************************************//* Arduino menu library example
Oct. 2016 Rui Azevedo (ruihfazevedo@gmail.com) www.r-site.net

Digital keypad (3 buttons) using the menu keyIn driver
*/

#include <Arduino.h>
#include <menu.h>
#include <menuIO/serialIn.h>
#include <menuIO/serialOut.h>
#include <menuIO/altKeyIn.h>
#include <menuIO/chainStream.h>
//#include <menuIO/chainStream.h>    // Core graphics library
#include <Adafruit_GFX.h>
#include <menuIO/adafruitGfxOut.h> // Hardware-specific library for ST77XX
#include <Adafruit_ST7789.h> // Hardware-specific library for ST7789
#include <Adafruit_ST77xx.h>
#include <SPI.h>

#if defined(ARDUINO_FEATHER_ESP32) // Feather Huzzah32
  #define gfx_CS         14
  #define gfx_RST        15
  #define gfx_DC         32

#elif defined(ESP8266)
  #define gfx_CS         4
  #define gfx_RST        16                                            
  #define gfx_DC         5

[... 160 lines cut ...]

void setup() {
  Serial.begin(115200);
  while(!Serial);
  pinMode(LEDPIN, OUTPUT);
  joystickBtns.begin();

SPI.begin();
  gfx.init(240,135,0);
  gfx.setRotation(3);
  gfx.setTextSize(textScale);//test scalling
  gfx.setTextWrap(false);
  gfx.fillScreen(ST77XX_BLACK);
  gfx.setTextColor(ST77XX_RED,ST77XX_BLACK);
  gfx.println("Menu 4.x test on GFX");
  delay(1000);

#define SOFT_DEBOUNCE_MS 100
}
void loop() {
  
 
  nav.poll();//also do  input
  digitalWrite(LEDPIN, ledCtrl);
}