neu-rah/ArduinoMenu #317: Need help with MENU_OUTPUTS

fishercat503, 2020-07-03:
I am having a lot of difficulties understanding how to use the ADAGFX_OUT macro as an argument for the MENU_OUTPUTS macro. I get a "type name not allowed" error. This is generated by the second line at the first parameter "gfx". I also believe I am missing a few entries at the end of the list in the list in parenthesis. I have tried looking through the documentation and the header files themselves. Any ideas?

  MENU_OUTPUTS(out,MAX_DEPTH
  ,`ADAGFX_OUT(gfx,colors,idx_t,plist)`
  ,SERIAL_OUT(Serial)
  ,NONE

Doug Robertson

neu-rah, 2020-07-03:
Hi!
be sure to include all needed file (from ada gfx example)
```
#include <Adafruit_GFX.h>
#include <menuIO/adafruitGfxOut.h>
//and some other, specific to your display to work with adafruit gfx
```

on the lines above it is expected that you have already created a graphic object for your display and named it gfx (can use other name but they must match)

example, creating a `gfx` object for **ST7755**:
```c++
Adafruit_ST7735 gfx(TFT_CS, TFT_DC, TFT_RST);
```

please add more details if you need more info (like the error output)

fishercat503, 2020-07-03:
I was hoping not to bury everyone with a bunch of code, but here it is.
As you can see I have used the needed includes and made the adjustments to use ST7789 instead of the ST7735.
I believe I have made the correct variable assignments.
Here is the expanded error code

{
	"resource": "/home/doug/Arduino/btnNav/btnNav/btnNav.ino",
	"owner": "C/C++",
	"severity": 8,
	"message": "type name is not allowed",
	"startLineNumber": 210,
	"startColumn": 1,
	"endLineNumber": 210,
	"endColumn": 13
} 

``/**************************************************************************
  This is a library for several Adafruit displays based on ST77* drivers.

  Works with the Adafruit 1.8" gfx Breakout w/SD card
    ----> http://www.adafruit.com/products/358
  The 1.8" gfx shield
    ----> https://www.adafruit.com/product/
  The 1.44" gfx breakout
    ----> https://www.adafruit.com/product/2088
  The 1.14" gfx breakout
  ----> https://www.adafruit.com/product/4383
  The 1.3" gfx breakout
  ----> https://www.adafruit.com/product/4313
  The 1.54" gfx breakout
    ----> https://www.adafruit.com/product/3787
  The 2.0" gfx breakout
    ----> https://www.adafruit.com/product/4311
  as well as Adafruit raw 1.8" gfx display
    ----> http://www.adafruit.com/products/618

  Check out the links above for our tutorials and wiring diagrams.
  These displays use SPI to communicate, 4 or 5 pins are required to
  interface (RST is optional).

 
  Written by Limor Fried/Ladyada for Adafruit Industries.
  MIT license, all text above must be included in any redistribution
  Adafruit invests time and resources providing this open source code,
  please support Adafruit and open-source hardware by purchasing
  products from Adafruit!
**************************************************************************//* Arduino menu library example
Oct. 2016 Rui Azevedo (ruihfazevedo@gmail.com) www.r-site.net

Digital keypad (3 buttons) using the menu keyIn driver
*/

#include <Arduino.h>
#include <menu.h>
#include <menuIO/serialIn.h>
#include <menuIO/serialOut.h>
#include <menuIO/altKeyIn.h>
#include <menuIO/chainStream.h>
#include <Adafruit_GFX.h>
#include <menuIO/adafruitGfxOut.h> // Hardware-specific library for ST77XX
#include <Adafruit_ST7789.h> // Hardware-specific library for ST7789
#include <Adafruit_ST77xx.h>
#include <SPI.h>
#include <baseMacros.h>

#if defined(ARDUINO_FEATHER_ESP32) // Feather Huzzah32
//  #define gfx_CS         14
  #define gfx_RST        15
  #define gfx_DC         32

[... 177 lines cut ...]
void setup() {
  Serial.begin(115200);
  while(!Serial);
  joystickBtns.begin();

SPI.begin();
 
#define SOFT_DEBOUNCE_MS 100
}
void loop() {

  gfx.init(240,135,0);
  gfx.setRotation(3);
  gfx.setTextSize(textScale);//test scalling
  gfx.setTextWrap(false);
  gfx.fillScreen(ST77XX_BLACK);
  gfx.setTextColor(ST77XX_RED,ST77XX_BLACK);
  gfx.println("Menu 4.x test on GFX");
  delay(1000);

  
 
  nav.poll();//also do  input
 // digitalWrite(LEDPIN, ledCtrl);
}

neu-rah, 2020-07-03:
can you point us line 210?

fishercat503, 2020-07-03:
Line 210 is the start of the MENU_OUTPUTS macro.  I've listed the whole macro below, I believe the line that throws the error is 211.

210  MENU_OUTPUTS(out,MAX_DEPTH
211 ,ADAGFX_OUT(gfx,colors,idx_t,plist,)
212  ,SERIAL_OUT(Serial)
213 ,NONE
214  );

neu-rah, 2020-07-03:
well, reviewing your code, `idx_t` is a type, so you need a value there, as a matter of fact you need 2 values for the font size there.

after that there is a problem with the color table, it is expected to have a constant size (6 lines)... and after that there is a problem with the color definitions like ST7755_xxxx, they seem to be missing, Of course you can replace those names with valid color values for your display.

here some code that compiles:
```c++
#include <Arduino.h>
#include <menu.h>
#include <menuIO/serialIn.h>
#include <menuIO/serialOut.h>
#include <menuIO/altKeyIn.h>
#include <menuIO/chainStream.h>
#include <Adafruit_GFX.h>
#include <menuIO/adafruitGfxOut.h> // Hardware-specific library for ST77XX
#include <Adafruit_ST7789.h> // Hardware-specific library for ST7789
#include <Adafruit_ST77xx.h>
#include <SPI.h>
#include <baseMacros.h>

#if defined(ARDUINO_FEATHER_ESP32) // Feather Huzzah32
//  #define gfx_CS         14
  #define gfx_RST        15
  #define gfx_DC         32

#elif defined(ESP8266)
  #define gfx_CS         4
  #define gfx_RST        16                                            
  #define gfx_DC         5

#else
  // For the breakout board, you can use any 2 or 3 pins.
  // These pins will also work for the 1.8" gfx shield.
  #define gfx_CS        10
  #define gfx_RST        9 // Or set to -1 and connect to Arduino RESET pin
  #define gfx_DC         8
#endif
   #define MAX_DEPTH 4
  #define textScale 1

                    
// OPTION 1 (recommended) is to use the HARDWARE SPI pins, which are unique
// to each board and not reassignable. For Arduino Uno: MOSI = pin 11 and
// SCLK = pin 13. This is the fastest mode of operation and is required if
// using the breakout board's microSD card.

// For 1.44" and 1.8" gfx with ST77XX use:
//Adafruit_ST77XX gfx = Adafruit_ST77XX(gfx_CS, gfx_DC, gfx_RST);

// For 1.14", 1.3", 1.54", and 2.0" gfx with ST7789:
Adafruit_ST7789 gfx = Adafruit_ST7789(gfx_CS, gfx_DC, gfx_RST);

// Color definitions
#define ST77XX_BLACK 0x0000
#define ST77XX_GRAY 0x8888
#define ST77XX_BLUE 0x001F
#define ST77XX_RED 0xF800
#define ST77XX_GREEN 0x07E0
#define ST77XX_CYAN 0x07FF
#define ST77XX_MAGENTA 0xF81F
#define ST77XX_YELLOW 0xFFE0
#define ST77XX_WHITE 0xFFFF

const colorDef<uint16_t> colors[6] MEMMODE={
  {{(uint16_t)ST77XX_BLACK,(uint16_t)ST77XX_BLACK}, {(uint16_t)ST77XX_BLACK, (uint16_t)ST77XX_BLUE,  (uint16_t)ST77XX_BLUE}},//bgColor
  {{(uint16_t)ST77XX_GRAY, (uint16_t)ST77XX_GRAY},  {(uint16_t)ST77XX_WHITE, (uint16_t)ST77XX_WHITE, (uint16_t)ST77XX_WHITE}},//fgColor
  {{(uint16_t)ST77XX_WHITE,(uint16_t)ST77XX_BLACK}, {(uint16_t)ST77XX_YELLOW,(uint16_t)ST77XX_YELLOW,(uint16_t)ST77XX_RED}},//valColor
  {{(uint16_t)ST77XX_WHITE,(uint16_t)ST77XX_BLACK}, {(uint16_t)ST77XX_WHITE, (uint16_t)ST77XX_YELLOW,(uint16_t)ST77XX_YELLOW}},//unitColor
  {{(uint16_t)ST77XX_WHITE,(uint16_t)ST77XX_GRAY},  {(uint16_t)ST77XX_BLACK, (uint16_t)ST77XX_BLUE,  (uint16_t)ST77XX_WHITE}},//cursorColor
  {{(uint16_t)ST77XX_WHITE,(uint16_t)ST77XX_YELLOW},{(uint16_t)ST77XX_BLUE,  (uint16_t)ST77XX_RED,   (uint16_t)ST77XX_RED}},//titleColor
};

using namespace Menu;

 
#define BTN_SEL -4  // Select button
#define BTN_UP -1 // Up
#define BTN_DOWN -0 // Down
 

result doAlert(eventMask e, prompt &item);

result showEvent(eventMask e) {
  gfx.print("event: ");
  gfx.println(e);
  return proceed;
}

int test=55;

result action1(eventMask e,navNode& nav, prompt &item) {
  gfx.print("action1 event:");
  gfx.println(e);
  gfx.flush();
  return proceed;
}

result action2(eventMask e) {
  gfx.print("actikcon2 event:");
  gfx.println(e);
  gfx.flush();
  return quit;
}
MENU(mainMenu,"Main menu",doNothing,noEvent,wrapStyle
  ,OP("Play",action1,anyEvent)
  ,OP("Setup",action2,enterEvent)
  //,SUBMENU(togOp)
  // ,FIELD(test,"Test","%",0,100,10,1,doNothing,noEvent,wrapStyle)
  // ,SUBMENU(subMenu)
 //  ,SUBMENU(setLed)
 // ,OP("LED On",myLedOn,enterEvent)
 // ,OP("LED Off",myLedOff,enterEvent)
 // ,SUBMENU(selMenu)
 // ,SUBMENU(chooseMenu)
  ,OP("Alert test",doAlert,enterEvent)
  ,EXIT("<Back")
);

keyMap joystickBtn_map[]={
 {-BTN_SEL, defaultNavCodes[enterCmd].ch,INPUT_PULLUP} ,
 {-BTN_UP, defaultNavCodes[upCmd].ch,INPUT_PULLUP} ,
 {-BTN_DOWN, defaultNavCodes[downCmd].ch,INPUT_PULLUP}  ,
};
keyIn<3> joystickBtns(joystickBtn_map);
const Menu::panel panels[] MEMMODE={{0,0,40,2}};
Menu::navNode* nodes[sizeof(panels)/sizeof(Menu::panel)];//navNodes to store navigation status
Menu::panelsList pList(panels,nodes,1);//a list of panels and nodes
Menu::idx_t serialTops[MAX_DEPTH];
Menu::serialOut outSerial(Serial,serialTops);
Menu::idx_t gfxTOPS[MAX_DEPTH];
// Menu::adaGfxOut (gfx)(gfxTOPS,pList);
// Menu::liquidCrystalOut outLCD(lcd,lcdTops,pList);
// Menu::adaGfxOut [1](gfx,colors,gfxTOPS,MEMMODE);
serialIn serial(serial);
menuIn* inputsList[]={&joystickBtns,&serial};
chainStream<2> in(inputsList);//3 is the number of inputs
// Menu::menuOut* const outputs[] MEMMODE=(adaGfxoutOut,&outSerial};
// Menu::outputsList out(gfx,2);//outputs list controller
MENU_OUTPUTS(out,MAX_DEPTH
  ,ADAGFX_OUT(gfx,colors,6,9,{0,0,14,8})
  ,SERIAL_OUT(Serial)
  ,NONE
  );

NAVROOT(nav,mainMenu,MAX_DEPTH,in,out);

result alert(menuOut& o,idleEvent e) {
  if (e==idling) {
    o.setCursor(0,0);
    o.print("alert test");
    o.setCursor(0,1);
    o.print("[select] to continue...");
  }
  return proceed;
}

result doAlert(eventMask e, prompt &item) {
  nav.idleOn(alert);
  return proceed;
}

void setup() {
  Serial.begin(115200);
  while(!Serial);
  joystickBtns.begin();

SPI.begin();
 
#define SOFT_DEBOUNCE_MS 100
}
void loop() {

  gfx.init(240,135,0);
  gfx.setRotation(3);
  gfx.setTextSize(textScale);//test scalling
  gfx.setTextWrap(false);
  gfx.fillScreen(ST77XX_BLACK);
  gfx.setTextColor(ST77XX_RED,ST77XX_BLACK);
  gfx.println("Menu 4.x test on GFX");
  delay(1000);

  
 
  nav.poll();//also do  input
 // digitalWrite(LEDPIN, ledCtrl);
}
```

fishercat503, 2020-07-04:
Thank you so much for your help. I think I can get this to work. As you can see, I do a lot of cut and paste. Then try to debug. I think I did pretty well to get this far.

closed 2020-07-04 by neu-rah