neu-rah/ArduinoMenu #322: assert do not compile with xtensa on platformio

pat1, 2020-08-25:
https://github.com/neu-rah/ArduinoMenu/blob/master/src/menuIo.h#L165

The use of "assert" in this library on MCU I think is problematic ...
But it do not compile for wemos on platformio:

xtensa-lx106-elf-g++ --version
xtensa-lx106-elf-g++ (GCC) 4.8.2

```
xtensa-lx106-elf-g++ -o .pio/build/d1_mini/src/measure.ino.cpp.o -c -fno-rtti -std=c++11 -fexceptions -w -Os -mlongcalls -mtext-section-literals -falign-functions=4 -U__STRICT_ANSI__ -ffunction-sections -fdata-sections -fno-exceptions -Wall -DPLATFORMIO=40304 -DESP8266 -DARDUINO_ARCH_ESP8266 -DARDUINO_ESP8266_WEMOS_D1MINI -DF_CPU=80000000L -D__ets__ -DICACHE_FLASH -DARDUINO=10805 -DARDUINO_BOARD=\"PLATFORMIO_D1_MINI\" -DFLASHMODE_DIO -DLWIP_OPEN_SRC -DNONOSDK22x_190703=1 -DTCP_MSS=536 -DLWIP_FEATURES=1 -DLWIP_IPV6=0 -DVTABLES_IN_FLASH -Iinclude -Isrc -I/home/pat1/git/rmap/arduino/sketchbook/libraries/ArduinoMenu/src -I/home/pat1/git/rmap/arduino/sketchbook/libraries/Rotary -I/home/pat1/git/rmap/arduino/sketchbook/libraries/streamFlow/src -I/home/pat1/git/rmap/arduino/sketchbook/libraries/ibt_2/src -I/home/pat1/git/rmap/arduino/sketchbook/libraries/WebSockets/src -I/home/pat1/.platformio/packages/framework-arduinoespressif8266/libraries/Hash/src -I/home/pat1/.platformio/packages/framework-arduinoespressif8266/libraries/Ethernet/src -I/home/pat1/git/rmap/arduino/sketchbook/libraries/U8g2/src -I/home/pat1/git/rmap/arduino/sketchbook/libraries/TimeAlarms -I/home/pat1/git/rmap/arduino/sketchbook/libraries/Time -I/home/pat1/git/rmap/arduino/sketchbook/libraries/SensorDriverb -I/home/pat1/.platformio/packages/framework-arduinoespressif8266/libraries/SPI -I/home/pat1/git/rmap/arduino/sketchbook/libraries/ArduinoJson/src -I/home/pat1/git/rmap/arduino/sketchbook/libraries/sps30/src -I/home/pat1/git/rmap/arduino/sketchbook/libraries/Registers -I/home/pat1/git/rmap/arduino/sketchbook/libraries/pms/src -I/home/pat1/git/rmap/arduino/sketchbook/libraries/hpm/src -I/home/pat1/git/rmap/arduino/sketchbook/libraries/EspSoftwareSerial/src -I/home/pat1/git/rmap/arduino/sketchbook/libraries/aJson -I/home/pat1/git/rmap/arduino/sketchbook/libraries/aJson/utility -I/home/pat1/git/rmap/arduino/sketchbook/libraries/SparkFun_SCD30_Arduino_Library/src -I/home/pat1/git/rmap/arduino/sketchbook/libraries/Sds011 -I/home/pat1/git/rmap/arduino/sketchbook/libraries/arduino-sht -I/home/pat1/.platformio/packages/framework-arduinoespressif8266/libraries/Wire -I/home/pat1/git/rmap/arduino/sketchbook/libraries/JC_Button/src -I/home/pat1/.platformio/packages/framework-arduinoespressif8266/libraries/DNSServer/src -I/home/pat1/git/rmap/arduino/sketchbook/libraries/ArduinoLog -I/home/pat1/git/rmap/arduino/sketchbook/libraries/circularbuffer -I/home/pat1/git/rmap/arduino/sketchbook/libraries/EspHtmlTemplateProcessor/src -I/home/pat1/.platformio/packages/framework-arduinoespressif8266/libraries/ESP8266WebServer/src -I/home/pat1/.platformio/packages/framework-arduinoespressif8266/libraries/ESP8266WiFi/src -I/home/pat1/git/rmap/arduino/sketchbook/libraries/Calibration -I/home/pat1/.platformio/packages/framework-arduinoespressif8266/tools/sdk/include -I/home/pat1/.platformio/packages/framework-arduinoespressif8266/tools/sdk/libc/xtensa-lx106-elf/include -I/home/pat1/.platformio/packages/framework-arduinoespressif8266/cores/esp8266 -I/home/pat1/.platformio/packages/framework-arduinoespressif8266/tools/sdk/lwip2/include -I/home/pat1/.platformio/packages/framework-arduinoespressif8266/variants/d1_mini src/measure.ino.cpp
In file included from /home/pat1/.platformio/packages/framework-arduinoespressif8266/libraries/ESP8266WebServer/src/detail/RequestHandler.h:6:0,
                 from /home/pat1/.platformio/packages/framework-arduinoespressif8266/libraries/ESP8266WebServer/src/ESP8266WebServer.h:70,
                 from /home/pat1/git/rmap/platformio/domotica/measure/src/measure.ino:105:
/home/pat1/git/rmap/arduino/sketchbook/libraries/ArduinoMenu/src/menuIo.h: In member function 'Menu::menuOut& Menu::outputsList::operator[](Menu::idx_t)':
/home/pat1/.platformio/packages/framework-arduinoespressif8266/tools/sdk/libc/xtensa-lx106-elf/include/assert.h:18:39: error: '__assert_func' was not declared in this scope
              __ASSERT_FUNC, PSTR(#__e)))
                                       ^
/home/pat1/git/rmap/arduino/sketchbook/libraries/ArduinoMenu/src/menuIo.h:165:11: note: in expansion of macro 'assert'
           assert(i<cnt);
           ^
/home/pat1/.platformio/packages/framework-arduinoespressif8266/tools/sdk/libc/xtensa-lx106-elf/include/assert.h:18:39: note: suggested alternative:
              __ASSERT_FUNC, PSTR(#__e)))
                                       ^
/home/pat1/git/rmap/arduino/sketchbook/libraries/ArduinoMenu/src/menuIo.h:165:11: note: in expansion of macro 'assert'
           assert(i<cnt);
           ^
In file included from /home/pat1/.platformio/packages/framework-arduinoespressif8266/tools/sdk/libc/xtensa-lx106-elf/include/stdlib.h:11:0,
                 from /home/pat1/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/Arduino.h:27,
                 from /tmp/tmp9b1fh6ct:1:
/home/pat1/.platformio/packages/framework-arduinoespressif8266/tools/sdk/libc/xtensa-lx106-elf/include/assert.h:42:13: note:   'esp8266webserver::__assert_func'
 void _EXFUN(__assert_func, (const char *, int, const char *, const char *)
             ^
/home/pat1/.platformio/packages/framework-arduinoespressif8266/tools/sdk/libc/xtensa-lx106-elf/include/_ansi.h:65:30: note: in definition of macro '_EXFUN'
 #define _EXFUN(name, proto)  name proto
                              ^
*** [.pio/build/d1_mini/src/measure.ino.cpp.o] Error 1

```

neu-rah, 2020-08-25:
you can remove/comment that line, its just a check

pat1, 2020-08-25:
yes, I have done it in my repo

neu-rah, 2020-08-25:
thanks for reporting them, I will give it a try with extensa and look for proper include... or remove the assert