neu-rah/ArduinoMenu #329: define two whole different menus at the same project

HamidSaffari, 2020-10-09:
Hi and best regard.
How we can define two complete menus and with different navigation and activate each one by two different keys for example?

olikraus, 2020-10-09:
By writing software or by using a full featured menu system.

neu-rah, 2020-10-09:
here an example of concurrent menus, https://github.com/neu-rah/ArduinoMenu/blob/master/examples/concurrent/concurrent/concurrent.ino

you can have multiple independent menus and switch between them or have them running at same time provided that they have independent IO, as @olikraus  pointed, it won't be automatic, you must either switch focus between them (calling poll) or do the proper setup for independent IO

HamidSaffari, 2020-10-10:
Thanks @neu-rah, your example (concurrent.ino) was completely helpful.

closed 2020-10-10 by HamidSaffari