neu-rah/ArduinoMenu #330: Dynamic menu (list) based on JSON

ClaesClaes, 2020-10-09:
Hi, I'm evaluating this lib for a project using an ESP32 and must say it's a fabulous timesaver and so easy to work with, thanks! 

One thing I have not understood yet is the possibility to build dynamic menus. I've looked at the dynamic example but not sure I can work with that.

So, here's what I'm looking for: Apart from using the menu to set parameters I would also like to be able to display a list of items imported from a JSON file stored on an SD card and be able to select one item with a callback to a menu function that can adjust the parameters for the selected item. Below is a pseudo JSON to illustrate ( the number of items can be growing up to possibly 100 or more):

    [{"fieldToSelect":"item1","id":"D65F","para1":25,"parar2":50,"flag":false},
    {"fieldToSelect":"item2","id":"BF49","para1":20,"parar2":50,"flag":true},
    {"fieldToSelect":"item3","id":"2DC3","para1":20,"parar2":52,"flag":false},
    {"fieldToSelect":"item4","id":"66CC","para1":25,"parar2":52,"flag":true},
    {"fieldToSelect":"item5","id":"A6C3","para1":20,"parar2":51,"flag":false}]

The menu should populate the list to display from the field "fieldToSelect". The user scrolls to select one of "item1" to "item5" which then opens a submenu where "para1", "para2" and "flag" can be adjusted.

Hope this makes sense. 

Is this something anyone has already implemented or could anyone give me some tips or advice on how I could implement this?

neu-rah, 2020-10-09:
Dynamic menus would be in the way of building a full menu structure using RAM, it works, but some alternatives are in place to avoid that memory waste, after all the data structure already exists, either on the for of array of in your case a json object... there are some specific cases for SD file system but it was added later a more generic structure that will just require one to print the entries, example here: https://github.com/neu-rah/ArduinoMenu/blob/master/src/plugin/userMenu.h

ClaesClaes, 2020-10-11:
Thanks, I'll probably need some time to see how I can use this. Thanks in the meantime!

ClaesClaes, 2020-10-13:
Hi, I'm very thankful for your time in helping me on this. I've now tested the example you provided. I think I get a general understanding on how I could get it to work for my purpose. However, the example plugin seems to be somehow limited to a maximum of 8 (dataSz=8) so called 'Targets'. Grateful for your comments on what I've experienced below.

Some observations: 
1) When I increase dataSz to 9, the output menu list 9 'Targets' but the last 'Exit' item (which should have been number 10) is no longer visible in the serial output. 
2) If I nonetheless try to type in 10 and enter. It simply opens the submenu for 'Target 1'.
3) It's not clear to me, without further looking into this, if this limitation is linked to the use of serial input or the custom menu plugin. Would it help if I choose another input (like a rotary encoder)?
4) dataSz is defined and initialised at compile time. I would need to do that during run time. I assume it would not be a problem but please comment.

neu-rah, 2020-10-13:
Hi, index accelerators are only 1 to 9, however you can use keys (numeric keypad) +-/* assuming you are using serial input.
Generally speaking, items beyond accel. 9 are only accessible by scroll.

ClaesClaes, 2020-10-21:
Thanks! I've decided to use a different approach for the menu for this project but will definitely consider this lib for any other project with static venues. Thanks again for your support.

closed 2020-10-21 by ClaesClaes