neu-rah/ArduinoMenu #332: Select betwen 300 items

PeterWay1, 2020-11-27:
Hello,
Tnx for great work.

I try to use this menu system in my aplication.
Everthing is ok...
But now i try to use larhe amount of items.
I have array like this "bread ..." " meat ... " etc...
At end i have 300 items.

Is possible to use this menus for selection betwèn 300 items?
Any example for this?

At first i like just simulate then i will add this items on Sd card.

I have 3 button keyboard, arduino with atmega128 and st7735 display.

Have a nice day..
Gregor

neu-rah, 2020-11-27:
Hi!

yes, it can be made so, please look at this example https://github.com/neu-rah/ArduinoMenu/tree/master/examples/targetSel

a menu with customized print function that given and index (by the menu system) will print from an array of items (or something else)

PeterWay1, 2020-11-27:
Tnx for help... this example is to latrge... i need only simple example---

Like this:
CHOOSE(target.chooseTest,chooseMenu,"Choose",doNothing,noEvent,noStyle
  ,VALUE("First",First,doNothing,noEvent)
  ,VALUE("Second",Second,doNothing,noEvent)
  ,VALUE("Third",Third,doNothing,noEvent)
  ,VALUE("Last",Last,doNothing,noEvent)
);

Is possible to use variable for name?
CHOOSE(target.chooseTest,chooseMenu,"Choose",doNothing,noEvent,noStyle
  ,VALUE(var1,1,doNothing,noEvent)
  ,VALUE(var2,2,doNothing,noEvent)
  ,VALUE(var3,3,doNothing,noEvent)
  ,VALUE(var4,4,doNothing,noEvent)
);

Tnx Greg

neu-rah, 2020-11-27:
what mcu are you using?

PeterWay1, 2020-11-28:
I use atmega128 ( with arduino support)
Or atme328 or 256..
Br Greg

neu-rah, 2020-11-28:
the atmega has not much memory, and by default it will use flash (so things get more complicated to access flash memory). The custom menu is the best way to conserve memory. Transposing all those items to populate a menu is a waste, custom menu will print from the array itself.

PeterWay1, 2020-11-28:
Yes, AVR have small RAM and FLASH. 
I have plane to add SD card (128 MB to 2 GB) and all data heva there.

For first step i need simple example with menus and dynamic items, after this i will try to optimaze code.