neu-rah/ArduinoMenu #338: Touch screen menu implementation

SimonBrazda, 2021-01-18:
Hi, I would like to use touch screen capabilities with your menu library, but I am struggling with the implementation. I would appreciate any help in this matter.

Hardware:
Arduino Due,
3.5inch SPI Module MSP3520 - http://www.lcdwiki.com/3.5inch_SPI_Module_ILI9488_SKU:MSP3520

So far I have managed to get TFT_eSPI library to work with the menu system as output stream and tested that the touch screen works on its own. I have browsed though all your examples and found this [example](https://github.com/neu-rah/ArduinoMenu/blob/master/examples/UTFT/UTFT/UTFT.ino) with a touch screen as input stream. So I have tried it, but no luck there. If I understand it correctly, URTouch library is meant to be used in conjunction with UTFT, which doesn't support my LCD.

What should I do? Should I create new wrapper input class for TFT_eSPI which would implement the menuIn abstract class? If so, How would i go about it? Because TFT_eSPI has the touch screen functions build into it, not like as with UTFT which has separate class for touch screen functionality, So I am a bit confused.

Thanks for any help.

My code so far:
```c++
/********************
Sept. 2014 Rui Azevedo - ruihfazevedo(@rrob@)gmail.com

menu output to standard arduino LCD (LiquidCrystal)
output: LCD
input: encoder and Serial
www.r-site.net
***/

#include <Arduino.h>
#include <SPI.h>
#include <SD.h>

#include <TFT_eSPI.h>
#include <ArduinoJson.h>
#include <menu.h>
// #include <menuIO/liquidCrystalOut.h>
#include <menuIO/serialOut.h>
#include <menuIO/TFT_eSPIOut.h>
#include <menuIO/utftOut.h>
#include <menuIO/chainStream.h>
#include <menuIO/urtouchIn.h>
#include <menuIO/serialIn.h>
#include <menuIO/keyIn.h>

#define LEDPIN LED_BUILTIN
#define MAX_DEPTH 2

#define CURRENT_INPUT_PIN A6
#define VOLTAGE_INPUT_PIN A5

const float reference_voltage = 4.57;
float calibration_voltage=0.51;

unsigned long last_measurement_time = 0;

#define Black RGB565(0,0,0)
#define Red	RGB565(255,0,0)
#define Green RGB565(0,255,0)
#define Blue RGB565(0,0,255)
#define Gray RGB565(128,128,128)
#define LighterRed RGB565(255,150,150)
#define LighterGreen RGB565(150,255,150)
#define LighterBlue RGB565(150,150,255)
#define DarkerRed RGB565(150,0,0)
#define DarkerGreen RGB565(0,150,0)
#define DarkerBlue RGB565(0,0,150)
#define Cyan RGB565(0,255,255)
#define Magenta RGB565(255,0,255)
#define Yellow RGB565(255,255,0)
#define White RGB565(255,255,255)

const colorDef<uint16_t> colors[6] MEMMODE={
  {{(uint16_t)Black,(uint16_t)Black}, {(uint16_t)Black, (uint16_t)Blue,  (uint16_t)Blue}},//bgColor
  {{(uint16_t)Gray, (uint16_t)Gray},  {(uint16_t)White, (uint16_t)White, (uint16_t)White}},//fgColor
  {{(uint16_t)White,(uint16_t)Black}, {(uint16_t)Yellow,(uint16_t)Yellow,(uint16_t)Red}},//valColor
[... 376 lines cut ...]
    //   joystickBtns.begin();

    tft.init();
    tft.setRotation(2);
    // gfx.setTextSize(textScale);//test scalling
    tft.setTextWrap(true);
    tft.setTextSize(4);
    tft.fillScreen(Black);
    tft.setTextColor(White, Black);

    nav.showTitle = false;
    nav.idleTask = measure;
    psu_init();
    nav.idleOn(measure);//this menu will start on idle state, press select to enter menu
}

bool blink(int timeOn,int timeOff) {return millis()%(unsigned long)(timeOn+timeOff)<(unsigned long)timeOn;}

#define SOFT_DEBOUNCE_MS 100

void loop() {
  nav.poll();
  digitalWrite(LEDPIN, blink(timeOn,timeOff));
}
```

SimonBrazda, 2021-01-29:
So I have managed to get the touchscreen to work with the menu system by creating new class inspired by [urtouch.h](https://github.com/neu-rah/ArduinoMenu/blob/master/src/menuIO/urtouchIn.h). The new class inherits from [menuIn](https://github.com/neu-rah/ArduinoMenu/blob/master/src/menuIo.h) and implements all the functionality in almost the same manner as [urtouch.h](https://github.com/neu-rah/ArduinoMenu/blob/master/src/menuIO/urtouchIn.h). The only difference is in the constructor where I instantiate _TFT_eSPI_ class as my input stream instead of _urtouch_ and few changes in _read_ function. I am assuming that this approach is very wasteful on resources because _TFT_eSPI_ has all the functionality for display output which is not needed in this case. Better approach would be to have separate class just for the touchscreen. I will look into this later.

The new class _TFT_eSPI_touchIn_:
```c++
#ifndef RSITE_ARDUINO_MENU_UTOUCH
  #define RSITE_ARDUINO_MENU_UTOUCH

  #include <menuIO/TFT_eSPIOut.h>
  #include <TFT_eSPI.h>
//   #include <URTouch.h>
  #include <menuDefs.h>

    namespace Menu {

        class TFT_eSPI_touchIn:public menuIn {
        public:
            // URTouch& touch;
            TFT_eSPI& touch;
            navRoot& root;
            TFT_eSPIOut& out;
            uint16_t startX,startY,z;
            int scrlY;
            bool touching;
            bool dragging;
            unsigned long evTime;
            TFT_eSPI_touchIn(TFT_eSPI& t,/*URTouch& t, */navRoot& root,TFT_eSPIOut& out):touch(t),root(root),out(out),touching(false),dragging(false) {}
            int available(void) {
                return touch.getTouch(&startX,&startY);
            }
            int peek(void) {return -1;}
            int read() {
                menuNode& m=root.active();
                panel p=out.panels[out.panels.cur];
                if (touch.getTouch(&startX,&startY)) {
                    if (root.sleepTask) return options->navCodes[enterCmd].ch;
                    evTime=millis();
                    // touch.getTouch(&startX,&startY);
                    z = touch.getTouchRawZ();
                    Serial.println("x: " + String(startX) + ", y: " + String(startY) + ", z: " + String(z));
                    // startX=touch.getX()-p.x*out.resX;
                    // Serial.println("x: " + String(touch.getX()));
                    if (startX>out.maxX()*out.resX) return -1;
                    int y=startY-p.y*out.resY;//-menuNode::activeNode->oy;
                    // Serial.println("y: " + String(touch.getY()));
                    if (y<0||y>out.maxY()*out.resY) return -1;
                    // Serial.println("x: " + String(startX) + ", y: " + String(y));
                    //within menu box
                    if (touching) {//might be dragging
                        Serial.println("touching");
                        int d=scrlY-y;
                        int ad=abs(d);
                        if (ad>(out.resY>>1)&&(ad<<1)>out.resY) {
                            dragging=true;//we consider it a drag
                            Serial.println("dragging");
                            scrlY-=(d>0?1:-1)*(out.resY/2);
                            //prompt* nf=root.navFocus;
                            /*if (nf->isMenu()&&!nf->isVariant()) {
                                MENU_DEBUG_touch<<"utouch scrolling "<<(d>0?"Up":"Down")<<endl;
                                return (d>0?options->navCodes[scrlUpCmd].ch:options->navCodes[scrlDownCmd].ch);
                            } else {*/
                                //MENU_DEBUG_touch<<"utouch moving "<<(d>0?"Up":"Down")<<endl;
                                return (d>0?options->navCodes[upCmd].ch:options->navCodes[downCmd].ch);
                            //}
                        }
                    } else {//start new touching
                        touching=true;
                        dragging=false;
                        startY=y;
                        scrlY=y;
                        evTime=millis();
[... 18 lines cut ...]
                    }
                }
                return -1;
            }
            void flush() {}
            size_t write(uint8_t v) {return 0;}
        };

    }//namespace Menu

#endif
```

And here are the changed bits in _main.cpp_:
```c++
idx_t eSpiTops[MAX_DEPTH]={0};
TFT_eSPIOut eSpiOut(tft,colors,eSpiTops,pList,charW,charH+1);

extern navRoot nav;
// URTouch utouch(TFT_SCLK, TOUCH_CS, TFT_MOSI, TFT_MISO, 2);
TFT_eSPI_touchIn touch(tft, nav, eSpiOut);
MENU_INPUTS(in,&inSerial,&touch/*&joystickBtns*/);
```

Right now i am having issues with the controls, when I select menu FIELD item by double clicking it, I can increase or decrease its rough value by swiping up or down, however I can't do fine adjustments by double clicking it again. Anyone has same issue?

franzSusemichel, 2023-01-02:
Hi,

have you managed to solve the problems? I want to use a ILI9341 with a teensy 4.1 and it works fine if I use the TFT_eSPIOut library - but no luck when combining it with ArduinoMenu library