neu-rah/ArduinoMenu #342: How to prevent editing for EDIT?

hjpSalo, 2021-01-28:
Hello!
First of all, thank you for your hard work with this library!
I am using EDITs as a part of my menu structure, and in some occasions I need to allow editing those strings.
In some occasions I just need to select a function from few EDIT menu items, but when I select the item using enterCmd, then editing starts. Is there any way to prevent entering to the edit mode?
Thank you very much in advance!

neu-rah, 2021-01-30:
Hi! let me see if I got it correct, you have some items that you need to edit but also want to use those items to call an action? is that it?

hjpSalo, 2021-01-30:
Hi!
Yes. My device has 6 buttons for navigating etc., and it would be nice to assign separate functions for '*'-button and for the button that is used to function as enterCmd. Or it might be good as well to have a boolean control variable that user could set to enable/disable entering to the editing mode. There may exist a solution for this, but I didn't find it in the documentation.
Thank you!

neu-rah, 2021-02-01:
check this on wiki pls
https://github.com/neu-rah/ArduinoMenu/wiki/Navigation#enum-navcmds

hjpSalo, 2021-02-03:
It seems to me the problem is that for the EDIT the '*' is same as enterCmd. When I need to edit content of the buffer, I use '*' in order to enter to the edit mode and that works fine. But when I need to run the action of the EDIT menu item, I would like to use enterCmd. How's that?

hjpSalo, 2021-02-03:
Sorry!

hjpSalo, 2021-02-03:
It seems to me the problem is that for the EDIT the * is same as enterCmd. When I need to edit content of the buffer, I use * in order to enter to the edit mode and that works fine. But when I need to run the action of the EDIT menu item, I would like to use enterCmd. How's that?

neu-rah, 2021-02-04:
yes, there is besisdes deriving the edit class (for most extreme cases) you can use event handlers & event masks to listen to item events and in the case of enter you can decide if it enter edit or not by returning quit or proceed

this examples is usefull for exploring events: https://github.com/neu-rah/ArduinoMenu/blob/master/examples/handlers/handlers/handlers.ino