neu-rah/ArduinoMenu #343: callback in menuNode

ferchinas, 2021-02-08:
Hi Rui:
I want to execute a function when entering a submenu.
But if I add the callback function in the definition of the menuNode, when pressing "enter" the function is executed but it does not enter the menu.

`menuNode & subMenu = * new menuNode ("SENSOR 0", sizeof (sensorData) / sizeof (prompt *), sensorData, (Menu::callback) sensorMenu_0_cb, enterEvent);`

If I remove the action and eventMask parameters, I can correctly enter the submenu

`menuNode & subMenu = * new menuNode ("SENSOR 1", sizeof (sensorData) / sizeof (prompt *), sensorData);`

I tried to see if MENU_DEBUG gave me any information, but no message appears.
Any idea what I may be doing wrong?

neu-rah, 2021-02-08:
hi!
is `sensorMenu_0_cb` returning `proceed`?

ferchinas, 2021-02-08:
You are the best!
Problem solved.
It seems that I have not made major changes in the menu for a long time and I had not seen that now the callback functions return a value.
Thank you very much

closed 2021-02-08 by ferchinas