neu-rah/ArduinoMenu #344: TFT_eSPI AND FreeFont

Koxx3, 2021-02-08:
[image]

[image]

Koxx3, 2021-02-08:
FIY, I started with this sample :
https://github.com/neu-rah/ArduinoMenu/blob/master/examples/adafruitGfx_eTFT/TFT_eSPI/ArduinoMenu_LilyGo_TTGO_T-display_demo/ArduinoMenu_LilyGo_TTGO_T-display_demo.ino

neu-rah, 2021-02-09:
Hi, please change https://github.com/neu-rah/ArduinoMenu/blob/master/examples/adafruitGfx_eTFT/TFT_eSPI/ArduinoMenu_LilyGo_TTGO_T-display_demo/ArduinoMenu_LilyGo_TTGO_T-display_demo.ino#L250

to match you font character size

Koxx3, 2021-02-12:
ok, it worked ;)
thanks !

closed 2021-02-12 by Koxx3