neu-rah/ArduinoMenu #361: Filter for filepicker

gilphilbert, 2021-08-04:
The filepicker is fantastic - just made my life infinitely easier!

One question though: I've read through the code quickly and it seems the answer is no, but does the filePicker have a filter? For example, could I allow it to show only files with a specific extension (`*.txt`) or fitting a specific name (`*log.txt*`)?

hicksan, 2021-08-04:
If this is the filePickMenu as used in SDCard.ino or SdFat.ino (example sketches) then I would be very interested to know how you got it to work because it doesn't for me. Thanks very much in advance.

gilphilbert, 2021-08-05:
Can you share some code you've tried? I'm using this with U8g2 and custom input on an ESP32 with an SD card on a shared SPI bus using `SD.h` (code from `SDCard.ino`) - and I've had no issues at all.

hicksan, 2021-08-05:
Here is SdFat01.ino as modified. I am using it with a basic Uno clone and an SD card - no u8g2 here at all in this instance - input and output via serial monitor only - and it doesn't work for me. I get garbage characters where the SUBMENU line should be, and the program crashes and restarts when any key is pressed. This is with pins 10, 11, 12 & 13 used for SPI + 5V and GND connections and USB as the basic power and PC connection. The same hardware configuration works perfectly with the standard SdFat.h library examples.

#include <SPI.h>
#include <SdFat.h>
#include <menu.h>
#include <menuIO/serialIO.h>
#include <plugin/SdFatMenu.h>
//enable this include if using esp8266
//#include <menuIO/esp8266Out.h>

using namespace Menu;

//esp8266 SS (wemos mini)
#define SDCARD_SS SS
//#define SDCARD_SS 9
SdFat sd;

//function to handle file select
// declared here and implemented bellow because we need
// to give it as event handler for `filePickMenu`
// and we also need to refer to `filePickMenu` inside the function
result filePick(eventMask event, navNode& nav, prompt &item);

SDMenuT<CachedFSO<SdFat,32>> filePickMenu(sd,"SD Card","/",filePick,enterEvent);

//implementing the handler here after filePick is defined...
result filePick(eventMask event, navNode& nav, prompt &item) {
  // switch(event) {//for now events are filtered only for enter, so we dont need this checking
  //   case enterCmd:
      if (nav.root->navFocus==(navTarget*)&filePickMenu) {
        Serial.println();
        Serial.print("selected file:");
        Serial.println(filePickMenu.selectedFile);
        Serial.print("from folder:");
        Serial.println(filePickMenu.selectedFolder);
      }
  //     break;
  // }
  return proceed;
}

#define MAX_DEPTH 2

MENU(mainMenu,"Main menu",doNothing,noEvent,wrapStyle
  //,OP("Something...",doNothing,noEvent)
  ,SUBMENU(filePickMenu)
  ,OP("Something else...",doNothing,noEvent)
  ,EXIT("<Back")
);

MENU_OUTPUTS(out,MAX_DEPTH
  ,SERIAL_OUT(Serial)
  ,NONE//must have 2 items at least
);

serialIn serial(Serial);
NAVROOT(nav,mainMenu,MAX_DEPTH,serial,out);

void setup() {
  pinMode(SDCARD_SS,OUTPUT);
  Serial.begin(115200);
  while (!Serial);
  Serial.println();
  Serial.print("Initializing SD card...");
  if (!sd.begin(SDCARD_SS, SD_SCK_MHZ(50))) {
    sd.initErrorHalt();
    serial.println("SD Card Error");
  }
  filePickMenu.begin();//need this after sd begin
  Serial.println("initialization done.");
  nav.useAccel=false;
}

constexpr int menuFPS=25;
unsigned long nextPool=0;
void loop() {
  unsigned long now=millis();
  if(now>=nextPool) { 
    nav.poll();
    nextPool=now+1000/menuFPS;
  }
}

hicksan, 2021-08-15:
I have switched from a Mega to an ESP8266, to avoid the AVR Progmem memory issues. But am now getting loads of compile errors about type File (from SdFat.h) not being recognized.