neu-rah/ArduinoMenu #363: SDFat.ino and SDCard.ino not working

hicksan, 2021-08-10:
I set up the basic demo for reading an SD card with ArduinoMenu but it didn't work - output garbage characters and crashed repeatedly. Passed on to friends with more experience in C++, who diagnosed that strings were being referred to as pointers when the menu was being printed, and string data was being written into program memory on the Arduino, causing it to crash. The same problem occurred with both SDCard.ino, using the SDCard.h library and SdFat.ino, using the SdFat.h library.

hicksan, 2021-08-10:
We finally found the cause.

These libraries both require a declaration of MENU_USERAM variable. This has the effect, in items.h, of converting menu node items from garbage pointers to text (lines 259+, within the class definition of menuNode). It is declared in macros.h, on line 4 at the top of the file. Both files are located in the src directory of the library.

Unfortunately in macros.h, as supplied, it is commented out. All that is required is to uncomment it (edit the file to remove the "//" at the beginning of the line. Doing that completely fixed the problem and the file runs perfectly now.

ACuppaJoes, 2021-08-14:
Any chance you or your more experienced friends could help me out? I am trying to create a menu for a midi controller but the wiki documentation is tough for me to follow