neu-rah/ArduinoMenu #365: QWERTY Text Fields

ACuppaJoes, 2021-08-17:
I was wondering if it were possible to implement QWERTY or Alphabetical text input. Instead of scrolling through till you get to your desired character, you can input text using a submenu which displays selectable keys. 

Similar to this video:
https://youtu.be/tpUkqxKpRVE?t=807

polygonfuture, 2021-09-16:
I was wondering something similar for FIELD objects "string" param.

For instance trying to use an extended ASCII character for a menu text:   

`char* arrowUp = 30;

MENU(mainMenu, "Settings", doNothing, noEvent, wrapStyle
     , FIELD(dwellTop, arrowUp, " s", 5, 30, 1, 1, adjustDwellTop, enterEvent, wrapStyle)
     , EXIT("<  Sleep")
    );`

ACuppaJoes, 2021-09-18:
I don't think he checks the issues section regularly. I still haven't received a response but I am going to play around with the Target example. I think it might help us resolve this issue