neu-rah/ArduinoMenu #367: Limit the number of items in the menu

Astar76, 2021-09-17:
Is there a limit in the library for the number of menu items used? I have an ESP-32 microcontroller and the menu already has many items. But when adding a new TOGGLE element, I get an error, but not at the compilation stage, namely when you enter the menu section. It seems to me that this is due to the limitation on the number of elements.
Here's an example TOGGLE:

```
// todo добавить сохранение
TOGGLE(pixelsValue, pixels, "Pixels: ", doNothing, noEvent, noStyle,
    VALUE("1", 1, doNothing, noEvent),
    VALUE("2", 2, doNothing, noEvent),
    VALUE("3", 3, doNothing, noEvent),
    VALUE("4", 4, doNothing, noEvent),
    VALUE("6", 6, doNothing, noEvent),
    VALUE("8", 8, doNothing, noEvent),
    VALUE("12", 12, doNothing, noEvent),
    VALUE("16", 16, doNothing, noEvent),
    VALUE("24", 24, doNothing, noEvent),
    VALUE("32", 32, doNothing, noEvent),
    VALUE("48", 48, doNothing, noEvent),
    VALUE("96", 96, doNothing, noEvent)
);
```

hicksan, 2021-09-17:
Quoting neu-rah from another issue:
"yes, there is a limit on the number of items, they can be 127 on AVR mcu's and 32767 on non AVR's."
A toggle should be a two-way choice really, so there may be some problem with the number of options you are toggling between. This looks more like a case for SELECT or CHOOSE menu items to me. Trying that may fix your problem.

Astar76, 2021-09-21:
Спасибо за ответ

closed 2021-09-21 by Astar76