neu-rah/ArduinoMenu #368: Show graphics on the display

Astar76, 2021-09-22:
Is it possible to display graphics on the display through Adafruit GFX? I have an OLED display SSD1306. The menu is displayed well, but there was a need to show the battery charge in the upper right corner, since there is a lot of free space.

DisDis, 2022-10-01:
Hi,
Try to use #include <menuIO/adafruitGfxOut.h>
and will change initialization

```
#define gfxWidth 128
#define gfxHeight 64
#define fontX 7
#define fontY 10
MENU_OUTPUTS(out, MAX_DEPTH, ADAGFX_OUT(display, colors, fontX, fontY, {0, 0, gfxWidth / fontX, gfxHeight / fontY}), SERIAL_OUT(Serial));
```

But I faced with issue incorrect draw menu item

martykube, 2023-01-12:
Hi,
I have a similar use case.  @neu-rah do you have any guidance?
Thanks!

neu-rah, 2023-01-25:
@martykube use a read only numeric field tied to a variable where the sensor read is stored, it will automatically update its value whenever the sensor read changes...

To use custom draw.. there is a project that monitors servos that does indeed an inline plot.

AM4 is line based, the plot is restricted to a single line... next version will not.

keldonin, 2023-03-22:
@neu-rah, I suppose you are referring to this sketch: `https://github.com/jarkman/ServoBox/blob/master/ServoBox.ino`, is that correct?