neu-rah/ArduinoMenu #372: Can it directly jump to a menu item through code control

be-engineer, 2021-11-10:
My application is to use ESP32 to control an LCD. There are two-level menus on it. 
[image]
In addition, there is an APP on the mobile phone for interactive control with ESP32. The mobile application does not have a menu like LCD, only some buttons.
I want to directly jump to a certain sub-menu to execute a function through the buttons on the APP. I wonder if I can jump directly to a certain menu item of the LCD through the code?
I tried to use code like **nav.doNav(navCmd(idxCmd, 0));** but I couldn't achieve the required function. I searched the source code and examples, but I didn't see any relevant examples.
What should I do?

be-engineer, 2021-11-27:
Can anybody help? I want to jump to some menuitem in code and run current menu function,how can I do?

hicksan, 2021-11-27:
I'm afraid Neu-rah appears to have stopped supporting this library, which is a pity because it is the best library out there. There are no direct examples if what you are doing, so you will need to master the library code and data structures first in order to do it.

I think I may do something similar to what you describe using the idle function to suspend the menu and do all my other stuff. So my idle function is quite large and has several different screen displays, depending on variables set in the menu.  In order to do this in reverse and jump to some specific location in the menu you would need to set the menu location to a different value before you resume. What you are trying to do is at the "advanced" level but I'm sure is possible.

Try looking at the targetSel, example, or fullIdle.

be-engineer, 2021-11-29:
Thks hicksan,
I use these code can do some function well by directly jump to some menu item now,but when I use lcd to return to up level menu, it can not display normally.
```
nav.useMenu(systemSubMenu);
nav.doNav(navCmd(idxCmd, 2)); 
nav.refresh();
```
I have to step by step to explore the source code。