neu-rah/ArduinoMenu #376: CustomPrompt example not updating

danielkucera, 2022-01-08:
The example says:
```
//this prompt will count seconds and update himself on the screen.
```
but it is not updating, It still shows `0`
Looks like the function `changed` is not called:
https://github.com/neu-rah/ArduinoMenu/blob/master/examples/CustomPrompt/CustomPrompt/CustomPrompt.ino#L40

Am I missing something?

danielkucera, 2022-01-08:
I've changed it to this and it works:
```
  virtual void clearChanged(const navNode &nav,const menuOut& out,bool sub) {
    dirty = false;
    t=millis()/1000;
    if (last!=t) {
      dirty = true;
    }
  }
```
Is that a correct way?

sirias52, 2022-08-08:
i have this problem

CustomPrompt:40:16: error: conflicting return type specified for 'virtual bool altPrompt::clearChanged(const Menu::navNode&, const Menu::menuOut&, bool)'
   virtual bool clearChanged(const navNode &nav,const menuOut& out,bool sub=true) {

neu-rah, 2022-08-09:
a pull request would be very welcomed :D