neu-rah/ArduinoMenu #384: Error with empty submenu

stmllr, 2022-03-17:
I created a menu with some submenus and one of the submenus `subMenuTuner` should be empty, having just one item to exit the submenu.

This is how the submenu looks like:
``` c
result handleTuner(eventMask event, navNode& nav, prompt &item);

MENU(subMenuTuner,"TUNER",handleTuner,(Menu::eventMask)(enterEvent|exitEvent),wrapStyle
  // add menu items here
  ,EXIT("<BACK")
);
```

However, I end with compiler error:
```
exit status 1
'subMenuTuner_text' was not declared in this scope
```
 If I add another item to the menu, e.g. `OP("foo",doNothing,noEvent)` compiler works fine.
 
 What would be an appropriate way to generate an empty submenus?

stmllr, 2022-03-17:
compiler output:
 ``` c
 ~/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino5/bin/avr-g++ -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -MMD -flto -mmcu=atmega4809 -DF_CPU=16000000L -DARDUINO=10819 -DARDUINO_AVR_NANO_EVERY -DARDUINO_ARCH_MEGAAVR -DAVR_NANO_4809_328MODE -DMILLIS_USE_TIMERB3 -DNO_EXTERNAL_I2C_PULLUP -I~/.arduino15/packages/arduino/hardware/megaavr/1.8.7/cores/arduino/api/deprecated -I~/.arduino15/packages/arduino/hardware/megaavr/1.8.7/cores/arduino -I~/.arduino15/packages/arduino/hardware/megaavr/1.8.7/variants/nona4809 -I~/.arduino15/packages/arduino/hardware/megaavr/1.8.7/libraries/EEPROM/src -I~/.arduino15/packages/arduino/hardware/megaavr/1.8.7/libraries/Wire/src -I~/Arduino/libraries/Adafruit_ADS1X15 -I~/Arduino/libraries/Adafruit_BusIO -I~/Arduino/libraries/Adafruit_MCP4728 -I~/.arduino15/packages/arduino/hardware/megaavr/1.8.7/libraries/SPI/src -I~/Arduino/libraries/MIDI_Library/src -I~/Arduino/libraries/ArduinoMenu_library/src -I~/Arduino/libraries/U8g2/src -I~/Arduino/libraries/RotaryEncoder/src -I~/Arduino/libraries/Bounce2/src -I~/Arduino/libraries/LinkedList /tmp/arduino_build_31940/sketch/main.ino.cpp -o /tmp/arduino_build_31940/sketch/main.ino.cpp.o
In file included from ~/Arduino/libraries/ArduinoMenu_library/src/menuBase.h:31:0,
                 from ~/Arduino/libraries/ArduinoMenu_library/src/menuDefs.h:12,
                 from ~/Arduino/libraries/ArduinoMenu_library/src/menu.h:15,
                 from ~/tool/src/main/main.ino:8:
~/Arduino/libraries/ArduinoMenu_library/src/macros.h:62:17: error: 'DECL_' does not name a type; did you mean 'DECL'?
 #define DECL(x) DECL_##x
                 ^
~/Arduino/libraries/ArduinoMenu_library/src/baseMacros.h:9:34: note: in expansion of macro 'DECL'
 #define FOR_EACH_1(what, x, ...) what(x)
                                  ^~~~
~/Arduino/libraries/ArduinoMenu_library/src/baseMacros.h:12:3: note: in expansion of macro 'FOR_EACH_1'
   FOR_EACH_1(what,  __VA_ARGS__)
   ^~~~~~~~~~
~/Arduino/libraries/ArduinoMenu_library/src/baseMacros.h:7:35: note: in expansion of macro 'FOR_EACH_2'
 #define CONCATENATE2(arg1, arg2)  arg1##arg2
                                   ^~~~
~/Arduino/libraries/ArduinoMenu_library/src/baseMacros.h:109:32: note: in expansion of macro 'FOR_EACH_'
 #define FOR_EACH(what, x, ...) FOR_EACH_(FOR_EACH_NARG(x, __VA_ARGS__), what, x, __VA_ARGS__)
                                ^~~~~~~~~
~/Arduino/libraries/ArduinoMenu_library/src/macros.h:195:3: note: in expansion of macro 'FOR_EACH'
   FOR_EACH(DECL,__VA_ARGS__)\
   ^~~~~~~~
~/Arduino/libraries/ArduinoMenu_library/src/macros.h:192:42: note: in expansion of macro 'altMENU'
 #define MENU(id,text,aFn,mask,style,...) altMENU(Menu::menu,id,text,aFn,mask,style,(Menu::_menuData|Menu::_canNav),__VA_ARGS__)
                                          ^~~~~~~
~/tool/src/main/tuner.h:3:1: note: in expansion of macro 'MENU'
 MENU(subMenuTuner,"EMPTY SUBMENU",handleTuner,(Menu::eventMask)(enterEvent|exitEvent),wrapStyle
 ^~~~
~/Arduino/libraries/ArduinoMenu_library/src/macros.h:63:16: error: 'DEF_' was not declared in this scope
 #define DEF(x) DEF_##x,
                ^
~/Arduino/libraries/ArduinoMenu_library/src/baseMacros.h:9:34: note: in expansion of macro 'DEF'
 #define FOR_EACH_1(what, x, ...) what(x)
                                  ^~~~
~/Arduino/libraries/ArduinoMenu_library/src/baseMacros.h:12:3: note: in expansion of macro 'FOR_EACH_1'
   FOR_EACH_1(what,  __VA_ARGS__)
   ^~~~~~~~~~
~/Arduino/libraries/ArduinoMenu_library/src/baseMacros.h:7:35: note: in expansion of macro 'FOR_EACH_2'
 #define CONCATENATE2(arg1, arg2)  arg1##arg2
                                   ^~~~
~/Arduino/libraries/ArduinoMenu_library/src/baseMacros.h:109:32: note: in expansion of macro 'FOR_EACH_'
 #define FOR_EACH(what, x, ...) FOR_EACH_(FOR_EACH_NARG(x, __VA_ARGS__), what, x, __VA_ARGS__)
                                ^~~~~~~~~
~/Arduino/libraries/ArduinoMenu_library/src/macros.h:198:5: note: in expansion of macro 'FOR_EACH'
     FOR_EACH(DEF,__VA_ARGS__)\
     ^~~~~~~~
~/Arduino/libraries/ArduinoMenu_library/src/macros.h:192:42: note: in expansion of macro 'altMENU'
 #define MENU(id,text,aFn,mask,style,...) altMENU(Menu::menu,id,text,aFn,mask,style,(Menu::_menuData|Menu::_canNav),__VA_ARGS__)
                                          ^~~~~~~
~/tool/src/main/tuner.h:3:1: note: in expansion of macro 'MENU'
 MENU(subMenuTuner,"EMPTY SUBMENU",handleTuner,(Menu::eventMask)(enterEvent|exitEvent),wrapStyle
 ^~~~
~/Arduino/libraries/ArduinoMenu_library/src/macros.h:63:16: note: suggested alternative: 'DEF'
 #define DEF(x) DEF_##x,
                ^
~/Arduino/libraries/ArduinoMenu_library/src/baseMacros.h:9:34: note: in expansion of macro 'DEF'
 #define FOR_EACH_1(what, x, ...) what(x)
                                  ^~~~
~/Arduino/libraries/ArduinoMenu_library/src/baseMacros.h:12:3: note: in expansion of macro 'FOR_EACH_1'
   FOR_EACH_1(what,  __VA_ARGS__)
   ^~~~~~~~~~
~/Arduino/libraries/ArduinoMenu_library/src/baseMacros.h:7:35: note: in expansion of macro 'FOR_EACH_2'
 #define CONCATENATE2(arg1, arg2)  arg1##arg2
                                   ^~~~
~/Arduino/libraries/ArduinoMenu_library/src/baseMacros.h:109:32: note: in expansion of macro 'FOR_EACH_'
 #define FOR_EACH(what, x, ...) FOR_EACH_(FOR_EACH_NARG(x, __VA_ARGS__), what, x, __VA_ARGS__)
[... cut ...]
 MENU(subMenuTuner,"EMPTY SUBMENU",handleTuner,(Menu::eventMask)(enterEvent|exitEvent),wrapStyle
 ^~~~
~/tool/src/main/tuner.h:3:6: note: suggested alternative: 'subMenuTuner_data'
 MENU(subMenuTuner,"EMPTY SUBMENU",handleTuner,(Menu::eventMask)(enterEvent|exitEvent),wrapStyle
      ^
~/Arduino/libraries/ArduinoMenu_library/src/macros.h:203:5: note: in definition of macro 'altMENU'
     id##_text,\
     ^~
~/tool/src/main/tuner.h:3:1: note: in expansion of macro 'MENU'
 MENU(subMenuTuner,"EMPTY SUBMENU",handleTuner,(Menu::eventMask)(enterEvent|exitEvent),wrapStyle
 ^~~~
 ```

neu-rah, 2022-08-04:
correct, macros require a menu with at least 2 elements.

sudokit, 2024-11-30:
I have a similar weird error with this code:
```c++
int selTest = 0;
SELECT(selTest, selMenu, "Select", doNothing, noEvent, noStyle,
  ,VALUE("Zero", 0, doNothing, noEvent),  
  ,VALUE("One", 1, doNothing, noEvent),  
  ,VALUE("Two", 2, doNothing, noEvent),  
)
```
results in compiler error of:
```
.pio/libdeps/esp32dev/ArduinoMenu library/src/macros.h:63:16: error: 'DEF_' was not declared in this scope
 #define DEF(x) DEF_##x,
                ^~~~
```

neu-rah, 2024-12-09:
can you try to remove last `,` on the SELECT params list?