neu-rah/ArduinoMenu #390: Is enterEvent valid for the top level?

ajs123, 2022-07-28:
I have the top level of a menu structure defined as follows:

`altMENU(altTitle, ampSetup, "SETUP", setupEntry, enterEvent, noStyle, (Menu::_menuData|Menu::_canNav),
    SUBMENU(volumeMenu),
    SUBMENU(remoteMenu),
    SUBMENU(labelMenu),
    OP("SAVE", saveValues, enterEvent),
    EXIT("<< EXIT")
    );
`

Defined this way, the callback setupEntry is never invoked (neither upon initial start nor when returning from the submenus). If I change the event flag to anyEvent, the callback does execute at various times. 

Right now, I'm invoking setupEntry prior to entering the menu and in the exitEvent callbacks for the submenus, so I do have a reasonable alternative. But I'm wondering whether I'm missing something.

Thanks!

neu-rah, 2022-07-29:
Hi!

you can try your define with the event print sketch https://github.com/neu-rah/ArduinoMenu/blob/725f4d56caf18bbaa219022b0b428a54a25bd638/examples/handlers/handlers/handlers.ino

ajs123, 2022-07-29:
Thanks.  I set showEvent as the callback...

`altMENU(altTitle,remoteMenu, "Remote", showEvent, anyEvent, noStyle,(Menu::_menuData|Menu::_canNav),
    OP("Vol +", volPlusOp, enterEvent),
    OP("Vol -", volMinusOp, enterEvent),
    OP("Mute", muteOp, enterEvent),
    OP("Input", inputOp, enterEvent),
    OP("Power", powerOp, enterEvent),
    OP("<< KEEP", saveRemoteCodesOp, enterEvent),
    OP("<< CANCEL", quitRemoteOp, enterEvent)
    );
`

Scrolling down through the menu gives focus and blur events as shown below. There's no event for entering this top-level menu, or for returning from any of the sub-menus. I guess that's expected behavior since when dropping into a sub-menu we're still in the top level menu. I also don't see an exitEvent when leaving the top level, and I guess that's normal since there's no actual exit from the top level, just the idle state. Am I understanding this correctly?

As I noted before, I'm using a separate function prior to initial entry, plus  the exitEvents of the sub-menus, so I do have a solution. I just want to be sure I'm not missing something!

========
Event for target: 0x20006F94
nav level:0 path:[1]
16 focusEvent

========
Event for target: 0x20006F94
nav level:0 path:[1]
32 blurEvent

neu-rah, 2022-08-03:
hi! :D
root level is indeed different, we use idle state for that level.

https://github.com/neu-rah/ArduinoMenu/wiki/Idling

ajs123, 2022-08-03:
Thanks, @neu-rah. As noted at the outset, I am handling tasks at return to the root level via the exit events of the submenus, and handling tasks at initial entry by a separate call prior to entry.

My use case, by the way, is implementation of a change indicator that is valid if the user changes any value, then changes it back to the original. It's all working very nicely!  Thanks again.

closed 2022-08-03 by ajs123

neu-rah, 2022-08-03:
you're welcome
you might want to take a look at `cancelField` example on plugins folder, it restores values on esc