neu-rah/ArduinoMenu #394: Large value numeric fields?

hicksan, 2022-08-26:
Might it be possible to have an additional form of the FIELD entry for menus, to handle inputs of large values.

At present we have:
FIELD - numeric field for editing and range validate a numeric variable (int, float, double, unsigned int, etc...)

FIELD(var.name, title, units, min., max., step size,fine step size, action, events mask, styles)

int pwm=0;
...
FIELD(pwm,"Power","%",0,100,10,1,doNothing,noEvent,noStyle)

This allows the value up to 100 to be incremented/decremented in steps of 10 or 1 - which is fine if the value is in the range 0-100, but less so if the range is 0-65,536 (a 2-byte value) or for any number with more than 3 significant digits. Inputting those might require lots (potentially hundreds or more) of button presses.

Yes - it could be input as a 4-digit HEX value but many client system end users wouldn't know how to convert decimal into HEX, so might not appreciate or understand that option.

Ideally the format for very large range numbers would simply progress through each digit from left to right setting each in turn, in steps of 1.

Might that be possible please? It would be very useful.

hicksan, 2022-12-18:
How do we input large numeric values using this system?

neu-rah, 2022-12-19:
hi, the `FIELD` uses the type of the variable (__pwm__) that you provide, it can be `long int` of `float` (i guess). The limits can be set anywhere on the type domain.

There are some field customization on the plugins folder, a field that uses acceleration instead of steps is more suitable to double/float data types, not coded that, but it can be added.

Beware that for each unique type provided, a new expansion of the template is done for that type... it can grow flash usage if using many different types. But still not related to the data type size.

hicksan, 2023-02-24:
Any chance this could be added please?

hicksan, 2023-02-24:
I have lots of inputs that are integers 4 to 6 digits long. How should I set them up?