neu-rah/ArduinoMenu #396: Status Bar

hicksan, 2022-09-13:
Is there an easy way to pre-allocate one row of the display as not part of the menu window - ie under direct user control? I would like to include a status bar with info from other parts of my system updating to the screen (eg time and other data from sensors). Effectively overwriting the bottom row of the menu and triggering a screen refresh when data updates (except when a menu field value is being edited, or that would be lost) but is there a more efficient way to do it?

hicksan, 2022-09-15:
Thanks Rui. That's great. Can you point me towards an example please?

neu-rah, 2022-09-15:
from the wiki https://github.com/neu-rah/ArduinoMenu/wiki/Output#panel

hicksan, 2022-09-15:
Thank you for this. An actual example would help, to show how 2 panels can be used with a menu in one and some text in the other - like a time counter or something - but I will try to start with this. Thanks.