neu-rah/ArduinoMenu #403: Failing to compile with MENU_ASYNC

bjonnh, 2022-12-18:
Trying to compile with MENU_ASYNC, I get the following errors:
```
.pio/libdeps/megaatmega2560/ArduinoMenu library/src/items.cpp: In member function 'virtual bool Menu::menuNode::async(const char*, Menu::navRoot&, Menu::idx_t)':
.pio/libdeps/megaatmega2560/ArduinoMenu library/src/items.cpp:99:37: error: 'class Menu::prompt' has no member named 'type'; did you mean 'style'?
     bool toggleUpdate=operator[](n).type()==toggleClass&&strchr(uri,'/');
                                     ^~~~
                                     style
.pio/libdeps/megaatmega2560/ArduinoMenu library/src/items.cpp:99:45: error: 'toggleClass' was not declared in this scope
     bool toggleUpdate=operator[](n).type()==toggleClass&&strchr(uri,'/');
                                             ^~~~~~~~~~~
.pio/libdeps/megaatmega2560/ArduinoMenu library/src/items.cpp:99:45: note: suggested alternative: 'toggleUpdate'
     bool toggleUpdate=operator[](n).type()==toggleClass&&strchr(uri,'/');
                                             ^~~~~~~~~~~
                                             toggleUpdate
```

These are the libraries I have and their versions
```
	fortyseveneffects/MIDI Library@^5.0.2
	arkhipenko/TaskScheduler@^3.7.0
	fmalpartida/LiquidCrystal@^1.5.0
	olikraus/U8g2@^2.34.8
	neu-rah/ArduinoMenu library@^4.21.4
	neu-rah/PCINT r-site.net@^4.0.8
	soligen2010/ClickEncoder@0.0.0-alpha+sha.9337a0c46c
	paulstoffregen/TimerOne@^1.1
	adafruit/Adafruit NeoPixel@^1.10.7
	neu-rah/streamFlow @ 0.0.0-alpha+sha.bf16ce8926
```

bjonnh, 2022-12-18:
I had to add: "#define MENU_FMT_WRAPS" as well for it to compile, but then the menus just completely crash and cannot be accessed at all. That's on a an arduino mega, I'm not sure if I would need to use RAM for that?

neu-rah, 2022-12-19:
only tested async on esp's, anyone used this on a mega?