neu-rah/ArduinoMenu #404: Support for ILI9341_t3

franzSusemichel, 2023-01-02:
Hi, 

I'm using the library with a teensy 4.1 and ST7920 display. Now i like to use it with a ILI9341 display. ILI9341_t3 library seems not be supported - but TFT_eSPI seems to  be fine. 

So i tested TFT_eSPI which worked fine - but no look with any of the TFT_eSPI examples in this library. I only get a black screen with blue bars. Increasing the font size increases the size of the blue bars but no luck with displaying the menu. 

Any hints for me? 

Kind regards

alioth78, 2023-01-03:
I am using an ILI9341 display but using the Adafruit Library. The problem is Adafruit library is very very slow in comparation with ili9341_t3.
And I am using some critical PIDs calculations that go wrong with refreshing menus. This didn't happen with my old nokia 5110 display.
I would like to use the ILI9341_t3 with my teensy too, which is really fast.

does anyone know how to use it?

Thanks.

franzSusemichel, 2023-01-03:
> I am using an ILI9341 display but using the Adafruit Library. The problem is Adafruit library is very very slow in comparation with ili9341_t3. And I am using some critical PIDs calculations that go [...]

Do you have a working example with adafruit display?

alioth78, 2023-01-03:
Here is a working example. I have tested:
[testILI01.zip](https://github.com/neu-rah/ArduinoMenu/files/10339806/testILI01.zip)

PINS, tested with teensy 4.1:
CS 10
DC 20
SDI 11
SCK 13
SDO 12
LED 24
encoderA 21
encoderB 22
encoder button 23

franzSusemichel, 2023-01-03:
Thx, do you know how i can figure out if led may be switched by teensy pin or needs an extra transistor?